ARMmbed/mbed-cli #501: mbed config <TOOLCHAIN_PATH> <PATH> doesn't work in mbed OS tree

pan-, 2017-06-09:
It is should be possible to set a local toolchain path for an mbed OS tree. Not just a "valid mbed program". 

In my use case, in CI server, a global toolchain is already set via `mbed config -G` and this value has priority over the environment variable` $MBED_<toolchain name>=<toolchain_path>`. 
Because the mbed OS tree is not considered as a "valid mbed program" I cannot compile tests with a toolchain different from the one globally set.

screamerbg, 2017-06-21:
@pan- A codebase is considered a "valid mbed program" when there is a `.mbed` configuration file in the root of the program/project. It's the same configuration file that holds the local config for that specific program that is set via `mbed config <toolchain>_path <path>`

Therefore if you initialize a program using `mbed config root .` in the root of the codebase, you would then be able to use `mbed config <toolchain>_path path` and be able to set the preference for the toolchain for that specific codebase.

theotherjimmy, 2017-11-28:
Please re-open this issue if your problem has not been resolved.

closed 2017-11-28 by theotherjimmy