ARMmbed/mbed-cli #502: mbed-os-example-blinky breaks on NRF52_DK when NDEBUG macro set

saedelman, 2017-06-11:
Both online compiler (using Compile Macros -> NDEBUG) and MBED-CLI (mbed_app.json) causes application freeze in blinky with NRF52_DK as target. I have not tested it for other targets.

bridadan, 2017-06-12:
@saedelman Thanks for reporting this. Could you please report this at https://github.com/ARMmbed/mbed-os? Could also include the version of mbed-os or the commit hash of the blinky example you're using when you report it there? Thanks!

screamerbg, 2017-06-21:
@saedelman I think this is a linked issue https://github.com/ARMmbed/mbed-os/issues/2467. As @bridadan suggested, please raise an issue against https://github.com/ARMmbed/mbed-os so the Nordic engineering team have visibility on this.

saedelman, 2017-06-23:
@bridadan @screamerbg Done. https://github.com/ARMmbed/mbed-os/issues/4621

closed 2017-06-23 by bridadan