ARMmbed/mbed-cli #503: OpenSUSE mbed-cli installation

dnvasta, 2017-06-11:
what should i do now!
When i do execute [  mbed --version ]
it gives following errors

```
Traceback (most recent call last):
  File "/usr/bin/mbed-cli", line 9, in <module>
    load_entry_point('mbed-cli==1.1.1', 'console_scripts', 'mbed-cli')()
  File "/usr/lib/python3.4/site-packages/pkg_resources/__init__.py", line 558, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/usr/lib/python3.4/site-packages/pkg_resources/__init__.py", line 2682, in load_entry_point
    return ep.load()
  File "/usr/lib/python3.4/site-packages/pkg_resources/__init__.py", line 2355, in load
    return self.resolve()
  File "/usr/lib/python3.4/site-packages/pkg_resources/__init__.py", line 2361, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/lib/python3.4/site-packages/mbed/mbed.py", line 1065
    sorted_scms = sorted(sorted_scms, key=lambda (m, _): not m)
                                                 ^
SyntaxError: invalid 
```

dnvasta, 2017-06-11:
[ python --version ]
Python 2.7.12

[ pip --version ]
pip 7.1.2 from /usr/lib/python3.4/site-packages (python 3.4)

[ pip list ]
apparmor (2.10.2)
bottle (0.12.8)
coverage (3.7.1)
cupshelpers (1.0)
LibAppArmor (2.10.2)
mbed-cli (1.1.1)
nose (1.3.7)
pip (7.1.2)
py (1.4.30)
pycups (1.9.72)
pycurl (7.19.5.1)
pygobject (3.20.1)
pysmbc (1.0.15.4)
requests (2.7.0)
setuptools (18.3.2)
simplejson (3.8.0)

theotherjimmy, 2017-06-12:
@dnvasta mbed CLI needs python 2 ATM, could you do:
```bash
pip uninstall mbed-cli
pip2 install mbed-cli
```

Long explination:

You installed mbed CLI with python3.4. mbed CLI does not run well with python3.4 (see your traceback). The only way I know to get it to run correctly is to install it with the python2 version of pip.

dnvasta, 2017-06-12:
@theotherjimmy Great! Its all good now! Thank you for your valuable reply!

closed 2017-06-12 by dnvasta