ARMmbed/mbed-cli #504: mbed export is missing parameter --app-config

AriParkkila, 2017-06-13:
For example _mbed-os-example-mesh-minimal_ need to be compiled with --app-config parameter. That parameter seems to be missing from mbed export command.

bridadan, 2017-06-13:
Thanks for reporting this @AriParkkila.

@theotherjimmy Do the tools support overriding the default filename `mbed_app.json` when exporting projects via the `--app-config` parameter?

theotherjimmy, 2017-06-13:
Let me check real quick.

theotherjimmy, 2017-06-13:
Not right now, but it would be a simple patch.

theotherjimmy, 2017-06-13:
```diff
diff --git i/tools/export/__init__.py w/tools/export/__init__.py
index e586c22b8..64dadf68f 100644
--- i/tools/export/__init__.py
+++ w/tools/export/__init__.py
@@ -255,7 +255,7 @@ def export_project(src_paths, export_path, target, ide, libraries_paths=None,
                    linker_script=None, notify=None, verbose=False, name=None,
                    inc_dirs=None, jobs=1, silent=False, extra_verbose=False,
                    config=None, macros=None, zip_proj=None, inc_repos=False,
-                   build_profile=None):
+                   build_profile=None, app_config=None):
     """Generates a project file and creates a zip archive if specified
 
     Positional Arguments:
@@ -307,7 +307,8 @@ def export_project(src_paths, export_path, target, ide, libraries_paths=None,
     toolchain = prepare_toolchain(
         paths, "", target, toolchain_name, macros=macros, jobs=jobs,
         notify=notify, silent=silent, verbose=verbose,
-        extra_verbose=extra_verbose, config=config, build_profile=build_profile)
+        extra_verbose=extra_verbose, config=config, build_profile=build_profile,
+        app_config=app_config)
     # The first path will give the name to the library
     if name is None:
         name = basename(normpath(abspath(src_paths[0])))
diff --git i/tools/project.py w/tools/project.py
index dce04ab64..d7185a78e 100644
--- i/tools/project.py
+++ w/tools/project.py
@@ -66,7 +66,8 @@ def setup_project(ide, target, program=None, source_dir=None, build=None, export
 
 
 def export(target, ide, build=None, src=None, macros=None, project_id=None,
-           zip_proj=False, build_profile=None, export_path=None, silent=False):
+           zip_proj=False, build_profile=None, export_path=None, silent=False,
+           app_config=None):
     """Do an export of a project.
 
     Positional arguments:
@@ -90,7 +91,8 @@ def export(target, ide, build=None, src=None, macros=None, project_id=None,
 
     return export_project(src, project_dir, target, ide, name=name,
                           macros=macros, libraries_paths=lib, zip_proj=zip_name,
-                          build_profile=build_profile, silent=silent)
+                          build_profile=build_profile, silent=silent,
+                          app_config=app_config)
 
 
 def main():
@@ -180,6 +182,9 @@ def main():
                         dest="update_packs",
                         action="store_true",
                         default=False)
+    pasres.add_argument("--app-config",
+                        dest="app_config",
+                        default=None)
 
     options = parser.parse_args()
 
@@ -245,7 +250,7 @@ def main():
     export(options.mcu, options.ide, build=options.build,
            src=options.source_dir, macros=options.macros,
            project_id=options.program, zip_proj=zip_proj,
-           build_profile=profile)
+           build_profile=profile, app_config=options.app_config)
 
 
 if __name__ == "__main__":
```

Would work

bridadan, 2017-06-13:
@theotherjimmy Lovely! Any chance you wanna bump up your PR count? 😄

theotherjimmy, 2017-06-13:
heh, see the linked PR.

bridadan, 2017-06-13:
Awesome, when that comes in we just need to add the option in mbed-cli as well.

theotherjimmy, 2017-06-13:
No we don't. mbed-cli just passes unknown arguments on untouched.

closed 2017-06-19 by theotherjimmy

bridadan, 2017-06-19:
@theotherjimmy We'll still need to add the mbed-cli option