ARMmbed/mbed-cli #507: Windows Uninstaller did not remove the git and mercurial which is installed by installer- enhancement

tz-arm, 2017-06-14:
The uninstaller did not remove git & mercurial which is installed by installer, propose to add a remove checkbox in uninstaller too.

theotherjimmy, 2017-11-28:
@arekzaluski Any news on this?

arekzaluski, 2017-12-01:
@theotherjimmy @tz-arm There is a ticket to implement this behaviour. ETA is not known yet.

theotherjimmy, 2017-12-01:
Thanks @arekzaluski. Could you keep this issue up to date? I would appreciate that.

screamerbg, 2018-03-23:
@arekzaluski What's the status of this?

arekzaluski, 2018-03-26:
@tz-arm @theotherjimmy @screamerbg This functionality has been implemented in newest version of installer : 0.5.1
Uninstaller removes now both github and mercurial if they were installed using installer.

screamerbg, 2018-05-26:
@arekzaluski Thank you!

closed 2018-05-26 by screamerbg