ARMmbed/mbed-cli #516: mbed import fails with OS Error when downloading from developper.mbed.org

adustm, 2017-06-27:
Hello,
I have an issue with mbed import / mbed deploy / mbed update
The following does not work
`mbed import -vv http://mbed.org/teams/ST/code/STM32_ADC_InternalChannels`
while the following works
`mbed import -vv mbed-os-example-blinky`
I am working on windows 7, and we have a proxy.
```
$ mbed import -vv http://mbed.org/teams/ST/code/STM32_ADC_InternalChannels
[mbed] Working path "C:\DEVELOPPER_MBED_ORG" (directory)
[mbed] Importing program "STM32_ADC_InternalChannels" from "https://mbed.org/teams/ST/code/STM32_ADC_InternalChannels" at latest revision in the current branch
[mbed] Exec "hg clone https://mbed.org/teams/ST/code/STM32_ADC_InternalChannels C:\DEVELOPPER_MBED_ORG\STM32_ADC_InternalChannels -v" in C:\DEVELOPPER_MBED_ORG
real URL is https://developer.mbed.org/teams/ST/code/STM32_ADC_InternalChannels/
requesting all changes
adding changesets
adding manifests
adding file changes
added 8 changesets with 9 changes to 2 files
updating to branch default
resolving manifests
getting main.cpp
getting mbed.bld
2 files updated, 0 files merged, 0 files removed, 0 files unresolved
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at rev #86740a56073b
[mbed] Setting url to "https://mbed.org/users/mbed_official/code/mbed/builds/tip" in C:\DEVELOPPER_MBED_ORG\STM32_ADC_InternalChannels\mbed
[mbed] Query "hg id --id https://mbed.org/users/mbed_official/code/mbed -r 86740a56073b" in C:\DEVELOPPER_MBED_ORG\STM32_ADC_InternalChannels\mbed
86740a56073b
[mbed] Downloading library build "86740a56073b" (might take a minute)
[mbed] ERROR: OS Error: The process cannot access the file because it is being used by another process
---
```
```
$ mbed -vv deploy
[mbed] Updating library "mbed" to rev #86740a56073b
[mbed] ERROR: An error occurred while unpacking library archive ".bld\.rev-86740a56073b.zip" in "C:\DEVELOPPER_MBED_ORG\STM32_ADC_InternalChannels\mbed"
```
It looks like my computer is able to download from github, but not from developper.mbed.org ? Or my computer is not able to unzip the build file ?

Tests done:
disable the firewall : still failing
disable the antivirus : still failing
connect to a home network (ie bypass the proxy) : I could not do that, because I did not find the correct way to change the settings when I have no proxy

It looks like my settings are correct to reach github but not to reach developper.mbed.org (but I am able to use hg commands to get examples and push examples...) 
I don't know how to solve that

adustm, 2017-06-27:
It looks like it is related to #390 #448

screamerbg, 2017-07-05:
I just imported the example fine with mbed CLI 1.2.

```
D:\1>mbed import https://developer.mbed.org/teams/ST/code/Nucleo_blink_led/
[mbed] Importing program "Nucleo_blink_led" from "https://developer.mbed.org/teams/ST/code/Nucleo_blink_led" at latest revision in the current branch
[mbed] Adding library "mbed" from "https://mbed.org/users/mbed_official/code/mbed/builds" at rev #86740a56073b

[mbed] Downloading library build "86740a56073b" (might take a minute)
[mbed] Unpacking library build "86740a56073b" in "D:\1\Nucleo_blink_led\mbed"
[mbed] Couldn't find build tools in your program. Downloading the mbed 2.0 SDK tools...
```

It's concerning that you get busy files, e.g.  `[mbed] ERROR: OS Error: The process cannot access the file because it is being used by another process`. Usually this is an indication of a anti-virus or other software that keeps the files busy.

Ruide, 2018-03-21:
Hi @adustm ,

I met the same issue. But after I restart my pc, and clear the cache files in .mbed folder, it works around. Hope this information helps you.

closed 2018-05-26 by screamerbg