ARMmbed/mbed-cli #517: export to uvision 5

noutram, 2017-06-27:
I have been trying to get the bluetooth HR demo to work with keil - https://github.com/ARMmbed/mbed-os-example-ble

Using a Nucleo F401 + IDB05A1 BLE board, I can confirm building with the CLI using `mbed compile` works fine - the HR example works as expected and I can view the data on an iPhone etc.

However....

When I export the project to uvision5 using  `mbed export -i uvision5` and build+deploy using uVision, the project builds+deploys, but does not function correctly (No advertised BT device + on board LED flashes too fast as well).

I am using the same ARM compiler (packaged with uVision) in both cases, yet there still seems to be a difference here. Is there a way to ensure they use the exact same compiler settings, or is there another issue I am not aware of?

theotherjimmy, 2017-06-27:
@noutram 

>Is there a way to ensure they use the exact same compiler settings?

You can get the full compiler command line from `mbed compile -v`

> or is there another issue I am not aware of?

I can't think of one.

Let me know how it goes!

noutram, 2017-06-27:
Good plan. I'll try that and report back!

theotherjimmy, 2017-06-27:
You could also check the `mbed_config.h` in `BUILD/NUCLEO_F401RE/ARM/` and compare it to the one in the project root.

theotherjimmy, 2017-06-27:
The first method verifies that the compiler command line is the same, the second verifies that the configuration is the same.

theotherjimmy, 2017-07-05:
@noutram Any news?

noutram, 2017-07-05:
There has been a patch to the libraries as I understand it (Not sure why this issue only flagged up with an export to uVision). As soon as I've tested it I will let you know

noutram, 2017-07-05:
https://github.com/ARMmbed/mbed-os-example-ble/issues/91

noutram, 2017-07-06:
The fix here https://github.com/ARMmbed/mbed-os-example-ble/issues/91#issuecomment-312590769 seems to have resolved this. I cannot claim to understand the issue entirely, or why it worked using the CLI and not with uVision. I'll now check the switches anyway

noutram, 2017-07-06:
mbed compile -v gives:

Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DMBED_BUILD_TIMESTAMP=1499336257.92 -DUSBHOST_OTHER -DTARGET_NUCLEO_F401RE -DTARGET_RTOS_M4_M7 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -DTARGET_STM32F4 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_I2C_ASYNCH=1 -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED -DFEATURE_BLE=1 -DDEVICE_PORTOUT=1 -DDEVICE_SPI_ASYNCH=1 -DTARGET_ST_BLUENRG -DDEVICE_SERIAL=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_STM32F401RE -DTARGET_FF_MORPHO -D__FPU_PRESENT=1 -DTARGET_FF_ARDUINO -DDEVICE_PORTIN=1 -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DUSB_STM_HAL -DDEVICE_ERROR_RED=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_PWMOUT=1 -DARM_MATH_CM4 -DTARGET_STM32F401xE

noutram, 2017-07-06:
@theotherjimmy mbed_config.h look identical

theotherjimmy, 2017-07-06:
@noutram Thanks for the debug. I have to point out that GCC_ARM compiler results are not useful for debugging uvision5, because it uses ARM.  Further, you should compare the Macros line to the defines in uvision. Any differences are generally a Bad Thing™.

noutram, 2017-07-06:
Understood and sorry - I need to get to my other machine to check this (where I had it all set up)

theotherjimmy, 2017-07-06:
Well, it seems that it has been resolved for you. That's good! I would like to know why it failed though :/

noutram, 2017-07-06:
Indeed - I'll do that as soon as I can.

noutram, 2017-07-07:
( @theotherjimmy  sorry for the delay - busy life + problems accessing our license server from the bus ;)

noutram, 2017-07-07:
uVision

**Include paths:**
.; .\DebugConfig; .\img; .\mbed-os; .\mbed-os\cmsis; .\mbed-os\cmsis\TARGET_CORTEX_M; .\mbed-os\cmsis\TARGET_CORTEX_M\TOOLCHAIN_ARM; .\mbed-os\drivers; .\mbed-os\events; .\mbed-os\events\equeue; .\mbed-os\features; .\mbed-os\features\filesystem; .\mbed-os\features\filesystem\bd; .\mbed-os\features\filesystem\fat; .\mbed-os\features\filesystem\fat\ChaN; .\mbed-os\features\frameworks; .\mbed-os\features\frameworks\greentea-client; .\mbed-os\features\frameworks\greentea-client\greentea-client; .\mbed-os\features\frameworks\greentea-client\source; .\mbed-os\features\frameworks\unity; .\mbed-os\features\frameworks\unity\source; .\mbed-os\features\frameworks\unity\unity; .\mbed-os\features\frameworks\utest; .\mbed-os\features\frameworks\utest\source; .\mbed-os\features\frameworks\utest\utest; .\mbed-os\features\nanostack; .\mbed-os\features\netsocket; .\mbed-os\features\netsocket\cellular; .\mbed-os\features\netsocket\cellular\generic_modem_driver; .\mbed-os\features\netsocket\cellular\utils; .\mbed-os\features\storage; .\mbed-os\hal; .\mbed-os\hal\storage_abstraction; .\mbed-os\platform; .\mbed-os\rtos; .\mbed-os\rtos\rtx4; .\mbed-os\rtos\rtx5; .\mbed-os\rtos\rtx5\TARGET_CORTEX_M; .\mbed-os\rtos\rtx5\TARGET_CORTEX_M\TARGET_RTOS_M4_M7; .\mbed-os\rtos\rtx5\TARGET_CORTEX_M\TARGET_RTOS_M4_M7\TOOLCHAIN_ARM; .\mbed-os\targets; .\mbed-os\targets\TARGET_STM; .\mbed-os\targets\TARGET_STM\TARGET_STM32F4; .\mbed-os\targets\TARGET_STM\TARGET_STM32F4\device; .\mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F401xE; .\mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F401xE\device; .\mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F401xE\device\TOOLCHAIN_ARM_STD; .\mbed-os\targets\TARGET_STM\TARGET_STM32F4\TARGET_STM32F401xE\TARGET_NUCLEO_F401RE; .\shields; .\shields\TARGET_ST_BLUENRG; .\shields\TARGET_ST_BLUENRG\bluenrg; .\shields\TARGET_ST_BLUENRG\bluenrg\bluenrg-hci; .\shields\TARGET_ST_BLUENRG\bluenrg\platform; .\shields\TARGET_ST_BLUENRG\bluenrg\utils; .\shields\TARGET_ST_BLUENRG\source; .\shields\TARGET_ST_BLUENRG\source\bluenrg-hci; .\shields\TARGET_ST_BLUENRG\source\bluenrg-hci\hci; .\shields\TARGET_ST_BLUENRG\source\bluenrg-hci\hci\controller; .\shields\TARGET_ST_BLUENRG\source\bluenrg-hci\utils; .\shields\TARGET_ST_BLUENRG\source\platform; .\shields\TARGET_ST_BLUENRG\source\utils; .\source; .\mbed-os\features\FEATURE_BLE; .\mbed-os\features\FEATURE_BLE\ble; .\mbed-os\features\FEATURE_BLE\ble\services; .\mbed-os\features\FEATURE_BLE\source; .\mbed-os\features\FEATURE_BLE\source\services; .\mbed-os\features\FEATURE_BLE\targets

**Misc Controls:**
-DDEVICE_RTC=1 -DTARGET_NUCLEO_F401RE -DDEVICE_SLEEP=1 -DTOOLCHAIN_object -DTOOLCHAIN_ARM_STD --preinclude=mbed_config.h --split_sections -DTARGET_STM32F401RE -DTARGET_STM32F4 -D__ASSERT_MSG --no_rtti -DMBED_BUILD_TIMESTAMP=1499412058.79 -DDEVICE_PORTINOUT=1 -DTARGET_ST_BLUENRG -DTARGET_FF_MORPHO -g -DDEVICE_I2C_ASYNCH=1 -DTARGET_FF_ARDUINO -c -DTARGET_RTOS_M4_M7 -DDEVICE_SPISLAVE=1 -DDEVICE_PORTOUT=1 -DDEVICE_STDIO_MESSAGES=1 -DTARGET_LIKE_MBED -DDEVICE_SERIAL_FC=1 --cpu=Cortex-M4.fp -DDEVICE_SERIAL=1 -DDEVICE_ANALOGIN=1 -DTARGET_LIKE_CORTEX_M4 -D__CORTEX_M4 -DDEVICE_ERROR_RED=1 -DTARGET_CORTEX_M -DARM_MATH_CM4 -DUSB_STM_HAL -DDEVICE_I2C=1 -DDEVICE_PORTIN=1 -DTARGET_STM -DTOOLCHAIN_ARM -DDEVICE_INTERRUPTIN=1 --no_depend_system_headers -DTARGET_UVISOR_UNSUPPORTED -DUSBHOST_OTHER --md -DDEVICE_PWMOUT=1 -DDEVICE_SERIAL_ASYNCH=1 --multibyte_chars --gnu --brief_diagnostics -DTARGET_DEBUG --apcs=interwork -DDEVICE_SPI=1 -D__MBED__=1 -DTARGET_STM32F401xE --restrict -DMBED_TRAP_ERRORS_ENABLED=1 -DTARGET_M4 -DMBED_DEBUG -D__FPU_PRESENT=1 -DDEVICE_I2CSLAVE=1 -D__CMSIS_RTOS -DDEVICE_SPI_ASYNCH=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DFEATURE_BLE=1 -D__MBED_CMSIS_RTOS_CM

**Compiler Control String**
--c99 -c --cpu Cortex-M4.fp -g -O0 --apcs=interwork -I . -I ./DebugConfig -I ./img -I ./mbed-os -I ./mbed-os/cmsis -I ./mbed-os/cmsis/TARGET_CORTEX_M -I ./mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_ARM -I ./mbed-os/drivers -I ./mbed-os/events -I ./mbed-os/events/equeue -I ./mbed-os/features -I ./mbed-os/features/filesystem -I ./mbed-os/features/filesystem/bd -I ./mbed-os/features/filesystem/fat -I ./mbed-os/features/filesystem/fat/ChaN -I ./mbed-os/features/frameworks -I ./mbed-os/features/frameworks/greentea-client -I ./mbed-os/features/frameworks/greentea-client/greentea-client -I ./mbed-os/features/frameworks/greentea-client/source -I ./mbed-os/features/frameworks/unity -I ./mbed-os/features/frameworks/unity/source -I ./mbed-os/features/frameworks/unity/unity -I ./mbed-os/features/frameworks/utest -I ./mbed-os/features/frameworks/utest/source -I ./mbed-os/features/frameworks/utest/utest -I ./mbed-os/features/nanostack -I ./mbed-os/features/netsocket -I ./mbed-os/features/netsocket/cellular -I ./mbed-os/features/netsocket/cellular/generic_modem_driver -I ./mbed-os/features/netsocket/cellular/utils -I ./mbed-os/features/storage -I ./mbed-os/hal -I ./mbed-os/hal/storage_abstraction -I ./mbed-os/platform -I ./mbed-os/rtos -I ./mbed-os/rtos/rtx4 -I ./mbed-os/rtos/rtx5 -I ./mbed-os/rtos/rtx5/TARGET_CORTEX_M -I ./mbed-os/rtos/rtx5/TARGET_CORTEX_M/TARGET_RTOS_M4_M7 -I ./mbed-os/rtos/rtx5/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_ARM -I ./mbed-os/targets -I ./mbed-os/targets/TARGET_STM -I ./mbed-os/targets/TARGET_STM/TARGET_STM32F4 -I ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/device -I ./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE - [... cut ...] strict -DMBED_TRAP_ERRORS_ENABLED=1 -DTARGET_M4 -DMBED_DEBUG -D__FPU_PRESENT=1 -DDEVICE_I2CSLAVE=1 -D__CMSIS_RTOS -DDEVICE_SPI_ASYNCH=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -DFEATURE_BLE=1 -D__MBED_CMSIS_RTOS_CM
-I./RTE/_BLE_HeartRate
-IC:/Keil_v5/ARM/PACK/Keil/STM32F4xx_DFP/2.11.0/Drivers/CMSIS/Device/ST/STM32F4xx/Include
-IC:/Keil_v5/ARM/CMSIS/Include
-IC:/Keil_v5/ARM/PACK/Keil/STM32F4xx_DFP/2.11.0
-D__UVISION_VERSION="523" -DSTM32F401xE
-o .\BUILD\*.o --omf_browse .\BUILD\*.crf --depend .\BUILD\*.d

theotherjimmy, 2017-07-07:
> -D__UVISION_VERSION="523"

Good to know that you are using the latest uvision.

This is not leading me to a conclusion. They look the same, except for ordering.

theotherjimmy, 2017-11-28:
I'm closing this as the original issue has been resolved.

closed 2017-11-28 by theotherjimmy