ARMmbed/mbed-cli #519: Transcluded README means 5.4 docs have 5.5 information

AnotherButler, 2017-06-27:
https://docs.mbed.com/docs/mbed-os-handbook/en/5.4/dev_tools/cli/#toolchain-selection says to use GCC 6 instead of GCC 4.9.3. This would be correct for 5.5, but this doc is for 5.4. I think this is because it's a transcluded page.

@iriark01 What's the best way to fix this for now?

iriark01, 2017-06-28:
mbed CLI needs to have branches for old versions.

AnotherButler, 2017-06-29:
@screamerbg Do you have branches available for older versions?

screamerbg, 2017-06-29:
You can point to the tags

AnotherButler, 2017-08-16:
@iriark01 Can you help me with this? My research so far has not been as productive as I would like.

iriark01, 2017-08-17:
If you know which version of CLI maps to which version of Mbed OS, you can in theory transclude pages from the release: https://github.com/ARMmbed/mbed-cli/releases

AnotherButler, 2017-08-23:
@iriark01 Thanks so much. :) We have now resolved this issue, so I am closing it. Please feel free to reopen if needed.

closed 2017-08-23 by AnotherButler