ARMmbed/mbed-cli #520: issue for git exclude

tz-arm, 2017-06-29:
Reproduce Flow:
git clone https://github.com/ARMmbed/ci-test-shield.git
mbed new .
>> here it add mbed-os to .git/info/exclude
.mbed deploy
**>> here it add more items but remove mbed-os which is not expected.

screamerbg, 2017-07-05:
@tz-arm I think there's a confusion of too many commands being used. E.g. you can either:
* `mbed import https://github.com/ARMmbed/ci-test-shield.git`; OR
* `git clone https://github.com/ARMmbed/ci-test-shield.git` and `mbed deploy`; OR

`mbed new` is to create a new program or a library.

tz-arm, 2017-07-11:
I assume there is such a use case to use mbed deploy after mbed new.
e.g. 
1. while I create a new program,  => mbed new 
2. add/import some library manually, create .lib file or copy .lib from other projects directly.
3. mbed deploy .

Could you correct me and share the right way for above use case?

theotherjimmy, 2017-11-28:
@tz-arm Your workflow works just fine in Mbed CLI 1.2.2:
```
# mbed new .
[mbed] Creating new program "test" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch latest
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#182bbd51bc8d1b6217477c2b5de3f1d0c5e6249f"
# echo "https://github.com/ARMmbed/mbed-littlefs#d02b3122f006aa201bca4efc699bae40971e5a00" > littlefs.lib
# mbed deploy .
[mbed] Updating library "mbed-os" to rev #182bbd51bc8d
[mbed] Adding library "littlefs" from "https://github.com/ARMmbed/mbed-littlefs" at rev #d02b3122f006
```

Are you complaining that this:
```
# git status
On branch master

No commits yet

Changes to be committed:
  (use "git rm --cached <file>..." to unstage)

	new file:   mbed-os.lib

Untracked files:
  (use "git add <file>..." to include in what will be committed)

	littlefs.lib
	littlefs/
	mbed-os/
# git diff 
```
includes both the littelfs and mbed-os directories?

Luka234, 2017-12-20:
I have also observed that git exclude file is not updated ok. If you do mbed deploy with some library folders already present in app then those libraries are not added to git exclude file (they are removed).  

For example:
   1. clone mbed application
   2. mbed deploy (in order to download used libraries)
         at this step all app libraries are added to app root folder and git exclude file (.git->info->exclude) in section **# subrepo ignores** contains all libraries -> that is expected and valid
   3. mbed deploy (this is simulation of scenario when we have fetched app changes from remote repo and, for example, new libraries are added)
         at this step libraries that where added at step 2 are just updated but during that procedure **# subrepo ignores** entries in exclude file are removed. -> invalid

screamerbg, 2018-05-26:
@Luka234 Can you add `mbed sync` after step 3 and let me know if this fixes your problem?

Luka234, 2018-06-15:
Yes, adding `mbed sync` step fixes exclude file.
Thx

closed 2018-06-18 by screamerbg