ARMmbed/mbed-cli #522: Compile builds with spaces in paths

noutram, 2017-07-06:
On Windows 10, I justy tried to build some code (known to work) and it failed. It turned out the issue was with there being a space in the path on that particular user account.

I assume this is a known issue, but I thought I should flag it. Maybe I should generate a PR for the READ_ME?

theotherjimmy, 2017-07-06:
Hey @noutram. We're fixing this issue. Let me get the PR number for you.

theotherjimmy, 2017-07-06:
Resolved by https://github.com/ARMmbed/mbed-os/pull/4468

theotherjimmy, 2017-07-06:
And merged :3

theotherjimmy, 2017-07-06:
Would you like a comment in the README for older versions of mbed OS?

noutram, 2017-07-11:
Might be an idea. It had me confused for quite a while (until, I confess, I read the error messages properly)

closed 2017-07-11 by noutram