ARMmbed/mbed-cli #523: mbed without RTOS

ismailhkose, 2017-07-13:
How can I build mbed without RTOS support, bare metal mbed?

The following code should generate very small binary.

[image]

However, text, data and bss sections are huge for such a simple program. How can I make it bare metal?
I only want to use board library. 
+----------------------+-------+-------+------+
| Module               | .text | .data | .bss |
+----------------------+-------+-------+------+
| Misc                 | 26357 |  2176 |  608 |
| hal                  |   636 |     0 |   24 |
| platform             |  1276 |     4 |  270 |
| targets/TARGET_Maxim |  4619 |   260 |  398 |
| Subtotals            | 32888 |  2440 | 1300 |
+----------------------+-------+-------+------+

sg-, 2017-07-28:
There are a few ways depending on what you're upto. You can use:
- the mbed 2 release library https://developer.mbed.org/users/mbed_official/code/mbed/
- the source version of the mbed 2 release library https://developer.mbed.org/users/mbed_official/code/mbed-dev/
- ignore the RTOS using the mbed OS development repo (obviously not a supported usecase) with .mbedignore https://docs.mbed.com/docs/mbed-os-handbook/en/latest/advanced/mbedignore/

theotherjimmy, 2017-11-28:
@ismailhkose Please reopen if the above did not answer your question

closed 2017-11-28 by theotherjimmy