ARMmbed/mbed-cli #525: Export for sw4cli - project does not build

noutram, 2017-07-26:
Hi

Is the exporter for System Workbench broken?

I tried exporting a mbed-os Blinky project using the CLI, exporting for sw4stm32
(I am using the GNU_ARM compiler and NUCLEO_F401. I can build the application on the command line with no issues.)

I think import the project into my workspace, and build. If my memory serves me correctly, this worked perfectly in the past.

Now when I try, (i) it takes a long time to build and (ii) there are compiler errors (see below)

Nick

make all 
Building file: ../mbed-os/tools/data/rpc/class.cpp
Invoking: MCU G++ Compiler
/Users/noutram/Documents/workspace/mbed-os-example-blinky/Debug
arm-none-eabi-g++ -mcpu=cortex-m4 -mthumb  -mfpu=fpv4-sp-d16 '-D__MBED__=1' '-DDEVICE_I2CSLAVE=1' '-D__FPU_PRESENT=1' '-DDEVICE_PORTOUT=1' -DUSBHOST_OTHER -DTARGET_NUCLEO_F401RE -DTARGET_RTOS_M4_M7 '-DDEVICE_RTC=1' -DTOOLCHAIN_object '-DDEVICE_SERIAL_ASYNCH=1' -DTARGET_STM32F4 -D__CMSIS_RTOS -DTOOLCHAIN_GCC '-DDEVICE_I2C_ASYNCH=1' -DTARGET_CORTEX_M -DTARGET_DEBUG -DTARGET_LIKE_CORTEX_M4 -DTARGET_M4 -DTARGET_UVISOR_UNSUPPORTED '-DMBED_BUILD_TIMESTAMP=1501095237.78' '-DDEVICE_SPI_ASYNCH=1' '-DDEVICE_SERIAL=1' '-DDEVICE_INTERRUPTIN=1' '-DDEVICE_I2C=1' '-DTRANSACTION_QUEUE_SIZE_SPI=2' -D__CORTEX_M4 '-DDEVICE_STDIO_MESSAGES=1' -DTARGET_STM32F401RE -DTARGET_FF_MORPHO -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO '-DDEVICE_PORTIN=1' -DTARGET_STM '-DDEVICE_SERIAL_FC=1' '-DDEVICE_PORTINOUT=1' -DTARGET_LIKE_MBED -D__MBED_CMSIS_RTOS_CM '-DDEVICE_SLEEP=1' -DTOOLCHAIN_GCC_ARM '-DDEVICE_SPI=1' -DUSB_STM_HAL '-DDEVICE_ERROR_RED=1' '-DDEVICE_SPISLAVE=1' '-DDEVICE_ANALOGIN=1' '-DDEVICE_PWMOUT=1' -DARM_MATH_CM4 -DTARGET_STM32F401xE -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/." -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./img" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/cmsis" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/cmsis/TARGET_CORTEX_M" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/cmsis/TARGET_CORTEX_M/TOOLCHAIN_GCC" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/drivers" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/events" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/events/equeue" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/filesystem" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/filesystem/bd" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/filesystem/fat" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/filesystem/fat/ChaN" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/greentea-client" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/greentea-client/greentea-client" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/greentea-client/source" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/unity" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/unity/source" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/unity/unity" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/utest" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/utest/source" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/frameworks/utest/utest" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/importer" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/inc" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/inc/mbedtls" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/platform" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/platform/inc" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/platform/src" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/src" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/targets" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/targets/TARGET_STM" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/mbedtls/targets/TARGET_STM/TARGET_STM32F4" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/nanostack" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/netsocket" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/netsocket/cellular" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/netsocket/cellular/generic_modem_driver" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/netsocket/cellular/utils" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/features/storage" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/hal" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/hal/storage_abstraction" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/platform" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/rtos" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/rtos/rtx4" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/rtos/rtx5" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/rtos/rtx5/TARGET_CORTEX_M" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/rtos/rtx5/TARGET_CORTEX_M/TARGET_RTOS_M4_M7" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/rtos/rtx5/TARGET_CORTEX_M/TARGET_RTOS_M4_M7/TOOLCHAIN_GCC" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets/TARGET_STM" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets/TARGET_STM/TARGET_STM32F4" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets/TARGET_STM/TARGET_STM32F4/device" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/device" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/device/TOOLCHAIN_GCC_ARM" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/./mbed-os/targets/TARGET_STM/TARGET_STM32F4/TARGET_STM32F401xE/TARGET_NUCLEO_F401RE" -I"/Users/noutram/Documents/workspace/mbed-os-example-blinky/" -O0 -g3 -Wall -fmessage-length=0 -ffunction-sections -c -fno-exceptions -fno-rtti -MMD -MP -MF"mbed-os/tools/data/rpc/class.d" -MT"mbed-os/tools/data/rpc/class.o" -o "mbed-os/tools/data/rpc/class.o" "../mbed-os/tools/data/rpc/class.cpp"
../mbed-os/tools/data/rpc/class.cpp:1:11: error: expected unqualified-id before '{' token
 class Rpc{{name}} : public RPC {
           ^
../mbed-os/tools/data/rpc/class.cpp:1:17: error: expected ';' after class definition
 class Rpc{{name}} : public RPC {
                 ^
../mbed-os/tools/data/rpc/class.cpp:1:19: error: expected unqualified-id before ':' token
 class Rpc{{name}} : public RPC {
                   ^
make: *** [mbed-os/tools/data/rpc/class.o] Error 1

theotherjimmy, 2017-07-26:
@noutram see armmbed/mbed-os#4779. TL;DR: Yes.

noutram, 2017-07-26:
Thanks. I'll try and find an older version where it still works until it's fixed.

Thanks for the (very) fast response.

closed 2017-07-26 by noutram

theotherjimmy, 2017-07-26:
@noutram Looking at the git history, I'm not sure you will be able to do that. I'm pretty sure the failure you're referring to has existed since I moved SW4STM32 into it's own dir, which according to git happened Jan 17. That's about 5.2 releases.

Instead, I recommend using the GNU ARM Eclipse exporter (`mbed export -i gnuarmeclispe`) or the eclipse_gcc_arm exporter.

theotherjimmy, 2017-07-26:
> Thanks for the (very) fast response.

Your welcome. It's part of my job ;)