ARMmbed/mbed-cli #527: mbed installer and mbed serial driver problem

rspelta, 2017-07-28:
Hello,
I'm using windows 7 and I tried to install mbed-cli using the installer named "mbed_installer_v041.exe"
When the installer tries to install the mbed serial driver I have already plugged to the PC the **ST NUCLEO_F429ZI** but it is not recognized from the installer:

[image]

I resolved this issue installing manually the driver using the file **en.stsw-link009.zip** provided by ST website.

Is there some advice in order to avoid to install manually this driver?

Thanks

bridadan, 2017-07-28:
@arekzaluski Do we have any plans to detect boards that are attached before installing the serial driver?

arekzaluski, 2017-08-01:
@rspelta @bridadan I'm able to reproduce it. It seems that mbedWinSerial does not support Nucleo F429ZI board. I'll investigate it further and check what can be done to improve the process.

rspelta, 2017-08-01:
Perfect, thank you!

bcostm, 2017-08-11:
Same problem with me using another board (NUCLEO_F103RB).

The STLINK driver must be installed first:
[[https://my.st.com/content/my_st_com/en/products/embedded-software/development-tool-software/stsw-link009.license%3d1502449373926.html]]

But running the mbed_installer_v041.exe after STLink driver is installed does not change the behaviour: same error message as described by @rspelta

screamerbg, 2017-09-15:
@arekzaluski please see above

screamerbg, 2018-05-26:
@arekzaluski @rspelta Is this still an issue?

closed 2018-06-18 by screamerbg