ARMmbed/mbed-cli #528: Update mbed OS tools to match requirements.txt

MarceloSalazar, 2017-08-03:
### Description
- Type: Enhancement
- Related issue: https://github.com/ARMmbed/mbed-ls/issues/188
- Priority:  Major / Minor

----------------------------------------------------------------
## Enhancement

When moving to a newer version of mbed OS (using `mbed update` or `mbed new` or `mbed import`), I'd expect the tools defined in [requirements.txt](https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.5/requirements.txt) to be automatically updated and benefit from new features been released (e.g. mbed-ls, greentea, etc).

It seems mbed CLI does not currently reinforce the installation of newer version of the tools, only checks whether the tools exists.

@bridadan raised as per your suggestion [here](https://github.com/ARMmbed/mbed-ls/issues/188#issuecomment-318748138)

MarceloSalazar, 2018-06-04:
This continues to be a problem when not having the latest mbed-ls version installed.
Another example: https://github.com/ARMmbed/mbed-os-example-cellular/issues/70

Again, Mbed CLI/OS should check that the latest versions of the required tools are installed, if not, attempt to install it.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2019-10-31 by ciarmcom