ARMmbed/mbed-cli #529: compile command generates build target directories case-sensitively

JammuKekkonen, 2017-08-04:
Although targets k64f and K64F are effectively the same, typing the targets in different cases causes new build target folder to be created and thus causes the recompilation of the whole project. Same applies to toolchain directory under the target directory.

korjaa, 2017-08-08:
Closed #530 as there are non-uppercase targets also.

korjaa, 2017-08-08:
@theotherjimmy, sorry didn't fully catch what you meant with the Mbed OS tools. I'm not familiar with the relation of mbed-cli with Mbed OS tools.

Was the idea that the platform name would be verified before compilation, but not touched in the mbed-cli phase?

theotherjimmy, 2017-08-08:
> I'm not familiar with the relation of mbed-cli with Mbed OS tools.

Simple Explanation:
mbed CLI:
  * target detection
  * Source control
  * Running tests
mbed OS tools:
  * Compiling
  * Exporting
  * Building tests

> Was the idea that the platform name would be verified before compilation, but not touched in the mbed-cli phase?

Actually, I was recommending that you just uppercase them without concern. The mbed OS tools will figure out what to build if you use the wrong case, so rely on that.

korjaa, 2017-08-09:
Thanks for that task split explanation.

About the uppercasing, wasn't that what the original PR was doing, just uppercasing without concern? Or is it that it was on the wrong place and should be at Mbed OS tools.

theotherjimmy, 2017-11-28:
For anyone that would like to help out: I think that upcasing the _directory names_ would be a path forward here. I don't currently have time, and I would like to see this changed.

screamerbg, 2018-06-22:
This is now available in Mbed CLI 1.7.3

closed 2018-06-22 by screamerbg