ARMmbed/mbed-cli #536: Mbed CLI option is missing to continue test suite execution even one test case fails

fahim-ublox, 2017-08-25:
### Problem:
We have test suites having multiple tests. If one test case fails, further test cases are not executed in a test suite. Mbed CLI option is missing to continue test suite execution even a test case fails.

### Expected response:
There should be option in mbed CLI  to continue test suite execution even on test failures.

### Observed response:
After a test failure, mbed test suite execution stops. We are unable to get complete results for reporting.

### Target 
UBLOX_C030_U201

### Example Logs of test cases in test suite:

- [1503573630.65][HTST][INF] host test detected: default_auto
- [1503573630.68][CONN][RXD] >>> Running 10 test cases...
- [1503573630.72][CONN][INF] found KV pair in stream: {{__testcase_name;Set randomise}}, queued...
- [1503573630.76][CONN][INF] found KV pair in stream: {{__testcase_name;UDP echo test}}, queued...
- [1503573630.79][CONN][INF] found KV pair in stream: {{__testcase_name;UDP recv sizes}}, queued...
- [1503573630.83][CONN][INF] found KV pair in stream: {{__testcase_name;UDP async echo test}}, queued...
- [1503573630.87][CONN][INF] found KV pair in stream: {{__testcase_name;TCP recv sizes}}, queued...
- [1503573630.91][CONN][INF] found KV pair in stream: {{__testcase_name;TCP async echo test}}, queued...
- [1503573630.95][CONN][INF] found KV pair in stream: {{__testcase_name;Alloc max sockets}}, queued...
- [1503573631.01][CONN][INF] found KV pair in stream: {{__testcase_name;Connect with credentials}}, queued...
- [1503573631.06][CONN][INF] found KV pair in stream: {{__testcase_name;Connect with preset credentials}}, queued...

### Executed test cases logs until a test case fails: 

- mbedgt: test suite 'ublox-at-cellular-interface-tests-unit_tests-default' ............................ FAIL in 225.80 sec
- 	test case: 'Set randomise' ................................................................... OK in 28.33 sec
- 	test case: 'TCP recv sizes' .................................................................. FAIL in 23.69 sec
- 	test case: 'UDP async echo test' ............................................................. OK in 75.80 sec
- 	test case: 'UDP echo test' ................................................................... OK in 32.91 sec
- 	test case: 'UDP recv sizes' .................................................................. OK in 48.45 sec
- mbedgt: checking for 'host_tests' directory above image directory structure

RobMeades, 2017-09-29:
Any feedback on this guys?  We are quite interested in such a feature.

bridadan, 2017-09-29:
This isn't a feature of mbed CLI but rather the way the test is written. The default way for the test to run is as you described. You can change this behavior by overriding the utest greentea abort handler: https://github.com/ARMmbed/mbed-os/blob/729ef153076ea9db4a9c9f81e90076960e035213/TESTS/mbed_drivers/echo/main.cpp#L44

However, I vaguely remember there being some issues where certain test failures weren't being reported correctly when this option was used, so I'd use with caution and run some sanity checks before using it too much. This issue was found quite sometime ago and I'm having trouble tracking it down so I'm not sure what the status of it is.

RobMeades, 2017-09-29:
The default way to write tests, as I understand it, is to use the assert mechanism, so I'd assumed that the mbed test system was, if you like, integrated with this mechanism so that we could influence the effect of the abort procedure from the command line.   If not I guess we'll have to try overriding the abort ourselves and see what happens.

screamerbg, 2018-05-26:
@RobMeades @fahim-ublox Did you find answer to your question, and do you think this need further discussion?

mudassar-ublox, 2018-05-31:
@screamerbg and @bridadan thanks, It worked for us by overriding "greentea_failure_handler" function.