ARMmbed/mbed-cli #537: Run test for the case that application contains a main function

maxpeng, 2017-08-25:
From the [mbed CLI documentation](https://docs.mbed.com/docs/mbed-os-handbook/en/latest/dev_tools/cli/#testing), it describes this:

> Note: mbed test does not work in applications that contain a main function that is outside of a TESTS directory.

For normal project layout, we will place a main function outside the TEST directory, and place unit test code inside TEST directory. I didn't want to rename the file containing main, so I tried to write the application C file containing main function as:
```
#ifndef UNIT_TEST
...
int main(void)
{
...
}
#endif
```
That is, when UNIT_TEST is defined, the application main will not be included. Then I tried to build the test code by this command:
```
mbed test -m xxx 0t xxx -DUNIT_TEST
```
Although the documentation does not describe it support `-D` option, the test code could be compiled successfully. Unfortunately, the `mbedgt` didn't like the `-D` option:
```
mbedgt: error: no such option: -D
[mbed] ERROR: "mbedgt" returned error code 2.
```

How do we build the application and test code easily by using mbed CLI?

bridadan, 2017-08-25:
@maxpeng That's a good work around!

You can do the build run step separately:

```
mbed test -m xxx -t xxx -DUNIT_TSET --compile
mbed test -m xxx -t xxx --run
```

maxpeng, 2017-08-25:
@bridadan Thanks for your tip - split the intended action into 2 steps.

I wonder if it is better for the `mbed test` command to automatically define some kind of macro to ease the task.

bridadan, 2018-05-29:
> I wonder if it is better for the mbed test command to automatically define some kind of macro to ease the task.

That seems reasonable honestly, and I bet a lot of folks would appreciate that.

@theotherjimmy how do you feel about that?

theotherjimmy, 2018-05-29:
Sounds like a workable plan to me.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

moshe-shahar, 2018-10-17:
I face similar issue, trying to build application (using `mbed compile`) and tests (using `mbed test`) under the same working tree. 

Please consider another way for the fix, by ignoring `main` functions outside of TESTS directories when building tests using _`mbed test`_.
In that way there is no need to use another compilation flag.

jenia81, 2018-10-17:
@screamerbg Fix of this issue will help us for compilation of our unit tests. I believe that also @TeroJaasko will find this useful (for compiling PAL unit tests from Pelion Client application). Is this something that can be considered to be added in next mbed-cli version?

theotherjimmy, 2018-10-17:
@jenia81 The Mbed CLI version has no bearing on the behavior of `mbed test --compile`, outside of accepting the `--icetea` and `--greentea` switches.

@moshe-shahar 
>by ignoring main functions outside of TESTS directories when building tests using mbed test.

I'm sure this would be useful, but _how_ do you propose we do that? As it stands, the build tools don't know where `main` lives, and only deal with source files.

moshe-shahar, 2018-10-18:
@theotherjimmy ,I understand.
Thanks for your answer.

j3hill, 2019-03-11:
@moshe-shahar 5 months have gone by, and it appears that Jimmy answered your question by explaining the limitations of what can be determined by the CLI when a program is being compiled. This issue will now be closed if there are no further questions.

moshe-shahar, 2019-03-11:
Ok, thanks!

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

bridadan, 2019-03-11:
Note that as of version 1.9.0, we add the `MBED_TEST_MODE` macro when compiling with `mbed test`. Here are the docs describing the feature: https://github.com/ARMmbed/mbed-os-5-docs/pull/1003

closed 2019-03-11 by ciarmcom