ARMmbed/mbed-cli #541: mbed detect should show firmware information

dhwalters423, 2017-09-08:
``mbed detect`` could be more useful if, in addition to the name of the target platform, it could also show the firmware version, a subset of what the ``mbedls`` command shows. For example, the current ``mbed detect`` output shows:

```
[mbed] Detected K64F, port COM4, mounted D:
[mbed] Supported toolchains for K64F
+--------+-----------+-----------+-----------+-----------+-----------+
| Target | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |
+--------+-----------+-----------+-----------+-----------+-----------+
| K64F   | Supported | Supported | Supported | Supported | Supported |
+--------+-----------+-----------+-----------+-----------+-----------+
Supported targets: 1
Supported toolchains: 3
```

It would be helpful to have at least one more column describing the DAPLink version, for example:
```
+--------+-----------+-----------+-----------+-----------+-----------+-----------------+
| Target | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    | daplink_version |
+--------+-----------+-----------+-----------+-----------+-----------+-----------------+
| K64F   | Supported | Supported | Supported | Supported | Supported | 0243            |
+--------+-----------+-----------+-----------+-----------+-----------+-----------------+
```
This could be useful information for debugging in particular. One case where it would have come in handy is I was having an issue with the DAPLink version being out of date, causing a crash when the .bin was dragged onto the board. I wasn't aware of the command ``mbedls`` and if I had the firmware information immediately available it could have saved me a bit of time and headache.

screamerbg, 2017-09-08:
Great suggestion, thank you! We accept patches ;)

dhwalters423, 2017-09-12:
@screamerbg Reference PR https://github.com/ARMmbed/mbed-os/pull/5077
Functionality was tested on 5.5 and will be tested on 5.6 rc when available. Looks like the underlying tools have not changed in 5.6 to affect the PR however

dhwalters423, 2017-10-12:
Merged https://github.com/ARMmbed/mbed-os/pull/5077

closed 2017-10-12 by dhwalters423