ARMmbed/mbed-cli #543: Config and Documentation for ARMC6 in mbed OS

theotherjimmy, 2017-09-13:
ARMC6 is an optional compiler in mbed OS. 

mbed CLI should mention how to install and configure ARMC6 in it's readme, and should include a method for defining `MBED_ARMC6_PATH` with global mbed CLI configuration. 

I think this first line, found here https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L1406, should be changed.
```python
        compilers = ['ARM', 'GCC_ARM', 'IAR']
        for c in compilers:
            if self.get_cfg(c+'_PATH'):
                env['MBED_'+c+'_PATH'] = self.get_cfg(c+'_PATH')
```

theotherjimmy, 2017-09-13:
Oh yeah, I'm going to do this, but just not today 😄

theotherjimmy, 2017-09-13:
@adbridge @screamerbg Is there a way to mirror this issue in jira?

adbridge, 2017-09-14:
@theotherjimmy  We should have this done for the actual release

adbridge, 2017-09-14:
@theotherjimmy If I know what Jira project mbed-cli issues go to then I can add this repo to the github2jira script.

theotherjimmy, 2017-09-14:
@adbridge I would just say that it could fall into the same bin as mbed OS, but I would ask Aaron first.

PR #544 (2017-09-15): Configure and document ARMC6 support
Resolves #543

@screamerbg could you review the code (First commit for your convenience)
@AnotherButler could you review the docs (Second commit for your convenience)
closed 2017-09-15 by screamerbg