ARMmbed/mbed-cli #547: Compiling static libs?

j3270, 2017-09-15:
Hello,

When compiling static libs using mbed-cli, any lib the references mbed-os, or the static version of bed-os (i.e pre-compiled), copies in all of the headers, linker script, etc. from mbed-os into the build directory of the static library.

This creates a problem when you have multiple libs that reference  mbed-os in your project and you want to use static builds of them vs re-compiling them every time.

I address this currently by building my static libs in a temporary location, and then just copy over the .h files and archive into LIBRARIES/NAME/ folder within the root of my project.  I delete the temporary build location.  I can then just compile the code I am working on instead of mbed-os, awesomelib1, awesomelib2,... etc. while working on the new code.

I am doing something wrong?  Does my question need more explanation?  My mbed-cli workspace structure follows;

mbed_cli_wkspc
---->libs
-------->mbed-os
-------->lib1
---->programs
--------->blinky
------------>BUILD
------------>LIBRARIES(static)
--------->blinky2

I even tried keeping the static copies in another location, without deleting the extra stuff that causes the issue, and still, same issue.  I might change to have my static builds in one location, so that I don't have multiple copies, one for each program, but that is a separate thought from this.

Eventually I want to look into adding mbed-cli to Sublime text using there build system features, but I need to learn that still.

Any suggestions, constructive comments on my workflow, etc. would be appreciated.

Thanks,

theotherjimmy, 2017-09-15:
>  I can then just compile the code I am working on instead of mbed-os, awesomelib1, awesomelib2,... etc. while working on the new code.

This will happen anyway, assuming that you do not request a clean build. the build tools keep track of what has changed (including headers) and recompile only what is needed.

If you don't have a copy of mbed OS in a project, mbed CLI will download the mbed 2 tools and place them in a hidden `.temp` directory inside each project. This is probably not what you want.

theotherjimmy, 2017-09-15:
There is a big disadvantage of using static libraries with mbed OS projects: You can't change the configuration.

j3270, 2017-09-15:
> This is probably not what you want.

Correct, I want my libs in a separate location that I maintain a single copy of.  I don't want new copies of libs I use frequently every time I create a new program, so I use --creat-only.

j3270, 2017-09-15:
Let me try something based on your comment and get back to you.

theotherjimmy, 2017-09-15:
> Let me try something based on your comment and get back to you.

Let me know how it goes. 😄 Maybe we can document how to work around this better as a result of your findings.

j3270, 2017-09-15:
@theotherjimmy 

So, I couldn't recall if I had tried building the first time with all the required --source paths, and then just trying, mbed compile on the next build, to see if the BUILD directory had everything I needed.  It doesn't.

What you said above is true if you have the libs in the root of your project, but not if you keep them separate.  So, if you keep them separate you must reference them every time, or build a static copy that you keep locally, which is what I do.

It would be cool if there was a flag that told mbed-cli to do the operation of just keeping the required .h files and archive in the library build directory when done building a library.  It isn't a big deal now that I am experimenting with scripting the process, but it took a bit to figure out.

Is my workflow irregular?  Isn't it common to just have one copy of your libraries source, under version control, that you can build static copies of and use?

For documentation purposes I can do a better job of writting up my process and share it with you if you think it would be helpful.  But if my process is irregular and not the norm, would it be helpful?

Thanks,

j3270, 2017-09-15:
Maybe I am making things harder on myself than need be.

If I think about it, having my libs in a remote location on my machine (that is under version control and available to others on a server) vs a remote location on a server that I pull from and place in the root of my project, isn't much different than if I compile static copies of the ones locally on my machine and then move the headers and archive over.  

The differences are;

1. If I add a library from the server using mbed-cli, it pulls in the source and additional stuff to work with hg/git.  Once compiled, its compiled and doesn't get recompiled unless I change it, or do a clean build.  The pro is, the lib is in the root of my project and I don't have to add its path when compiling every time.  The con is, I have multiple copies of the lib all over my workspace, all under version control.  Not sure if this is a con, or a personal problem.

2. If I stay with my current set up, it is basically the same thing, I have my libs under version control and available to others through hg/mbed or GitHub, but I have to manage moving the required files from the static build into the root of my project if I don't want to add the path every time, which is the current con.  The pro, I have one copy of the src under version control, on my machine.

I will have to think about this more...

screamerbg, 2018-05-26:
@j3270 Did you find answer to your question? Are you happy to close this issue if resolved for you?

closed 2018-06-18 by screamerbg