ARMmbed/mbed-cli #548: mbed cli not detecting toolchain

normads, 2017-09-25:
```
mbed compile -t ARM -m K64F
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS]
               [-S [{matrix,toolchains,targets}]] [-f GENERAL_FILTER_REGEX]
               [--stats-depth STATS_DEPTH] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rpc] [--usb] [--dsp]
               [--testlib] [--build-data BUILD_DATA] [-l LINKER_SCRIPT]
make.py: error: Could not find executable for ARM.
Currently set search path: No path set
[mbed] ERROR: "/usr/bin/python" returned error code 2.
[mbed] ERROR: Command "/usr/bin/python -u /Users/MalithY/Documents/test-project/mbed-os/tools/make.py -t ARM -m K64F --source . --build ./BUILD/K64F/ARM" in "/Users/MalithY/Documents/test-project"
```

tried setting path using mbed config

```
mbed config --list
[mbed] Global config:
GCC_ARM_PATH=/usr/local/Cellar/arm-gcc-bin/6-2017-q2-update/bin/

[mbed] Local config (/Users/MalithY/Documents/test-project):
No local configuration is set
```

tried setting the MBED_GCC_ARM_PATH variable

```
env | grep ARM_PATH
MBED_GCC_ARM_PATH=GCC_ARM_PATH=/usr/local/Cellar/arm-gcc-bin/6-2017-q2-update/bin/
```

tried setting in the mbed_settings.py

```
cat mbed_settings.py | grep ARM_PATH
#ARM_PATH = "C:/Program Files/ARM"
GCC_ARM_PATH = "/usr/local/Cellar/arm-gcc-bin/6-2017-q2-update/bin/"
```

compiler is available

```
/usr/local/Cellar/arm-gcc-bin/6-2017-q2-update/bin/arm-none-eabi-gcc
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.
```

theotherjimmy, 2017-09-25:
@normads The `ARM` toolchain uses the `ARM_PATH` configuration variable. You have set the variable used by the `GCC_ARM` toolchain.

normads, 2017-09-25:
@theotherjimmy I think I'm using the gcc toolchain arm-none-eabi-**gcc**

theotherjimmy, 2017-09-25:
@normads Then use `mbed compile -t GCC_ARM -m K64F` as your compile command.

normads, 2017-09-25:
@theotherjimmy silly me :) Thanks!

theotherjimmy, 2017-09-25:
NP. Could you close this issue if it's solved then?

closed 2017-09-25 by normads