ARMmbed/mbed-cli #552: Allow cache location to be set on the command line

mark-edgeworth, 2017-10-05:
I am attempting to use mbed cli in a sandbox installation as a part of another tool. I wish to use 'mbed import' to create a local copy of a remote repo, but this downloads mbed-os every time unless the cache is turned on which takes ages. I have tried the following:

1.  Turn the 'local' cache setting on. 
 - This fails since I have no project yet and there is nowhere to store the setting.
2. mbed new, set the cache and then then mbed import.
 -  the import fails because the import directory already exists.
3. Set the global cache setting.
 - This works, but changes the cache for the user's environment and hence affects any use of mbed cli outside of my sandbox.

Request for a command line option that overrides the cache setting for the duration of that command only. Something like 'mbed --config cache /path/to/cache import ...'. There are probably other setting that will need to be overridden in this way so allowing multiple '--config var value' entries might be a good plan.

screamerbg, 2017-10-05:
@mark-edgeworth This is infact already supported (but perhaps not well explained in the documentation). You can use `mbed config cache /path/to/cache --global`

mark-edgeworth, 2017-10-05:
Hi Mihail,

I’m currently using something like this to set the global cache to point at my preloaded cache; this works pretty well. However it changes a setting in the user’s home directory which then applies from that point onwards for any use of the tool either inside or outside my environment. I want to keep my changes completely private to avoid any user surprises (in case any users are already using mbed cli). Hence the request to set the config just for the duration of the import.
Ta,
Mark

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

theotherjimmy, 2017-11-28:
Given how long this has been open, I think that outside assistance would be appreciated.

screamerbg, 2018-03-23:
This is now supported in Mbed CLI 1.5.0. Thanks for the great suggestions!

```
$ mbed cache --help
usage: mbed cache [-h] [-v] [-vv] [on] [off] [dir] [ls] [purge]

Repository cache management
To minimize traffic and reduce import times, Mbed CLI can cache repositories by storing their indexes.
By default repository caching is turned on. Turn it off if you experience any problems.

positional arguments:
  on                   Turn repository caching on. Will use either the default
                       or the user specified cache directory.
  off                  Turn repository caching off. Note that this doesn't
                       purge cached repositories. See "purge".
  dir                  Set cache directory. Set to "default" to let mbed CLI
                       determine the cache directory location
                       (/Users/mihsto01/.mbed/mbed-cache/).
  ls                   List cached repositories and their sizes.
  purge                Purge cached repositories. Note that this doesn't turn
                       caching off

optional arguments:
  -h, --help           show this help message and exit
  -v, --verbose        Verbose diagnostic output
  -vv, --very_verbose  Very verbose diagnostic output
```

closed 2018-03-23 by screamerbg