ARMmbed/mbed-cli #554: Compile ERROR with target: NUCLEO_L073RZ

GiacomoThror, 2017-10-29:
I am trying to compile the standard Blinky Example for mbed OS 5 for the **NUCLEO_L073RZ**:
This is the configuration of my sytsem:

1. Ubuntu 16.04
2. Python 2.7.12
3. pip 9.0.1
4. I use virtualenv
5. mbed-cli 1.2.2

This is the steps that I did:

```
mkvirtualenv mbed
pip install mbed-cli
cd mbed-os-example-blinky/
mbed toolchain GCC_ARM
mbed target NUCLEO_L073RZ
mbed compile --clean -j 1 -vv
```

Start log:
```
[mbed] Working path "/home/giacomo/software/project/smartcooler/test-mbed5/test/mbed-os-example-blinky" (program)
[mbed] Exec "/home/giacomo/software/workspaces/virtualenv/mbed/bin/python -u /home/giacomo/software/project/smartcooler/test-mbed5/test/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_L073RZ --source . --build ./BUILD/NUCLEO_L073RZ/GCC_ARM -c -v -j 1" in /home/giacomo/software/project/smartcooler/test-mbed5/test/mbed-os-example-blinky
Building project mbed-os-example-blinky (NUCLEO_L073RZ, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
[DEBUG] Macros: -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_FF_MORPHO -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DDEVICE_STDIO_MESSAGES=1 -DTARGET_STM32L0 -DTARGET_CORTEX_M -DTARGET_M0P -DDEVICE_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNCH=1 -DTARGET_NUCLEO_L073RZ -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M0PLUS -DMBED_BUILD_TIMESTAMP=1509298301.41 -DDEVICE_PWMOUT=1 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DTARGET_STM32L073xx -DDEVICE_TRNG=1 -DARM_MATH_CM0PLUS -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DTARGET_STM32L073RZ -DDEVICE_FLASH=1 -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M0
Compile [  0.3%]: main.cpp
```
Log of the **error**:
```
Compile [ 63.2%]: rtx_evflags.c
[DEBUG] Compile: arm-none-eabi-gcc -std=gnu99 -c -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -fmessage-length=0 -fno-exceptions -fno-builtin -ffunction-sections -fdata-sections -funsigned-char -MMD -fno-delete-null-pointer-checks -fomit-frame-pointer -Os -mcpu=cortex-m0plus -mthumb -DDEVICE_FLASH=1 -D__MBED__=1 -DDEVICE_I2CSLAVE=1 -DTARGET_LIKE_MBED -DDEVICE_PORTOUT=1 -DDEVICE_PORTINOUT=1 -D__MBED_CMSIS_RTOS_CM -DTARGET_FF_MORPHO -DDEVICE_LOWPOWERTIMER=1 -DDEVICE_RTC=1 -DTOOLCHAIN_object -DDEVICE_SERIAL_ASYNCH=1 -D__CMSIS_RTOS -DTOOLCHAIN_GCC -DTARGET_STM32L0 -DTARGET_CORTEX_M -DTARGET_M0P -DDEVICE_ANALOGOUT=1 -DTARGET_UVISOR_UNSUPPORTED -DDEVICE_SPI_ASYNCH=1 -DTARGET_NUCLEO_L073RZ -DDEVICE_INTERRUPTIN=1 -DTARGET_CORTEX -DDEVICE_I2C=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M0PLUS -DDEVICE_STDIO_MESSAGES=1 -DDEVICE_PWMOUT=1 -DTARGET_FAMILY_STM32 -DTARGET_FF_ARDUINO -DTARGET_RELEASE -DTARGET_STM -DDEVICE_SERIAL_FC=1 -DTARGET_STM32L073xx -DDEVICE_TRNG=1 -DARM_MATH_CM0PLUS -DDEVICE_PORTIN=1 -DDEVICE_SLEEP=1 -DTOOLCHAIN_GCC_ARM -DDEVICE_SPI=1 -DDEVICE_SPISLAVE=1 -DDEVICE_ANALOGIN=1 -DDEVICE_SERIAL=1 -DTARGET_STM32L073RZ -DMBED_BUILD_TIMESTAMP=1509293999.68 -DDEVICE_I2C_ASYNCH=1 -DTARGET_LIKE_CORTEX_M0 @./BUILD/NUCLEO_L073RZ/GCC_ARM/.includes_06a08b6cefd9940b3e42dc54ae7a78b8.txt -include ./BUILD/NUCLEO_L073RZ/GCC_ARM/mbed_config.h -MD -MF ./BUILD/NUCLEO_L073RZ/GCC_ARM/mbed-os/rtos/TARGET_CORTEX/rtx5/rtx_evflags.d -o ./BUILD/NUCLEO_L073RZ/GCC_ARM/mbed-os/rtos/TARGET_CORTEX/rtx5/rtx_evflags.o ./mbed-os/rtos/TARGET_CORTEX/rtx5/rtx_evflags.c
[Error] ccatZFLR.s@677,0: cannot honor width suffix -- `mov r1,#0'
[Error] ccatZFLR.s@679,0: cannot honor width suffix -- `mov r0,#0'
[Error] ccatZFLR.s@727,0: cannot honor width suffix -- `mov r1,#4'
[Error] ccatZFLR.s@729,0: cannot honor width suffix -- `neg r1,r1'
[Error] ccatZFLR.s@731,0: cannot honor width suffix -- `mov r5,#4'
[Error] ccatZFLR.s@737,0: cannot honor width suffix -- `mov r1,#3'
[Error] ccatZFLR.s@739,0: cannot honor width suffix -- `neg r1,r1'
[Error] ccatZFLR.s@741,0: cannot honor width suffix -- `mov r5,#3'
[Error] ccatZFLR.s@743,0: cannot honor width suffix -- `neg r5,r5'
[Error] ccatZFLR.s@754,0: cannot honor width suffix -- `orr r5,r2'
[Error] ccatZFLR.s@904,0: cannot honor width suffix -- `mov r1,#4'
[Error] ccatZFLR.s@906,0: cannot honor width suffix -- `neg r1,r1'
[Error] ccatZFLR.s@908,0: cannot honor width suffix -- `mov r0,#4'
[Error] ccatZFLR.s@915,0: cannot honor width suffix -- `mov r1,#3'
[Error] ccatZFLR.s@916,0: cannot honor width suffix -- `neg r1,r1'
[Error] ccatZFLR.s@919,0: cannot honor width suffix -- `mov r0,#3'
[Error] ccatZFLR.s@921,0: cannot honor width suffix -- `neg r0,r0'
[Error] ccatZFLR.s@926,0: lo register required -- `add r0,r0,#12'
[Error] ccatZFLR.s@982,0: cannot honor width suffix -- `mov r1,#6'
[Error] ccatZFLR.s@984,0: cannot honor width suffix -- `neg r1,r1'
[Error] ccatZFLR.s@986,0: cannot honor width suffix -- `mov r0,#6'
[Error] ccatZFLR.s@987,0: cannot honor width suffix -- `neg r0,r0'
[DEBUG] Return: 1
[DEBUG] Output: /tmp/ccatZFLR.s: Assembler messages:
[DEBUG] Output: /tmp/ccatZFLR.s:677: Error: cannot honor width suffix -- `mov r1,#0'
[DEBUG] Output: /tmp/ccatZFLR.s:679: Error: cannot honor width suffix -- `mov r0,#0'
[DEBUG] Output: /tmp/ccatZFLR.s:727: Error: cannot honor width suffix -- `mov r1,#4'
[DEBUG] Output: /tmp/ccatZFLR.s:729: Error: cannot honor width suffix -- `neg r1,r1'
[DEBUG] Output: /tmp/ccatZFLR.s:731: Error: cannot honor width suffix -- `mov r5,#4'
[DEBUG] Output: /tmp/ccatZFLR.s:737: Error: cannot honor width suffix -- `mov r1,#3'
[DEBUG] Output: /tmp/ccatZFLR.s:739: Error: cannot honor width suffix -- `neg r1,r1'
[DEBUG] Output: /tmp/ccatZFLR.s:741: Error: cannot honor width suffix -- `mov r5,#3'
[DEBUG] Output: /tmp/ccatZFLR.s:743: Error: cannot honor width suffix -- `neg r5,r5'
[DEBUG] Output: /tmp/ccatZFLR.s:754: Error: cannot honor width suffix -- `orr r5,r2'
[DEBUG] Output: /tmp/ccatZFLR.s:904: Error: cannot honor width suffix -- `mov r1,#4'
[... 30 lines cut ...]
/tmp/ccatZFLR.s:741: Error: cannot honor width suffix -- `mov r5,#3'
/tmp/ccatZFLR.s:743: Error: cannot honor width suffix -- `neg r5,r5'
/tmp/ccatZFLR.s:754: Error: cannot honor width suffix -- `orr r5,r2'
/tmp/ccatZFLR.s:904: Error: cannot honor width suffix -- `mov r1,#4'
/tmp/ccatZFLR.s:906: Error: cannot honor width suffix -- `neg r1,r1'
/tmp/ccatZFLR.s:908: Error: cannot honor width suffix -- `mov r0,#4'
/tmp/ccatZFLR.s:915: Error: cannot honor width suffix -- `mov r1,#3'
/tmp/ccatZFLR.s:916: Error: cannot honor width suffix -- `neg r1,r1'
/tmp/ccatZFLR.s:919: Error: cannot honor width suffix -- `mov r0,#3'
/tmp/ccatZFLR.s:921: Error: cannot honor width suffix -- `neg r0,r0'
/tmp/ccatZFLR.s:926: Error: lo register required -- `add r0,r0,#12'
/tmp/ccatZFLR.s:982: Error: cannot honor width suffix -- `mov r1,#6'
/tmp/ccatZFLR.s:984: Error: cannot honor width suffix -- `neg r1,r1'
/tmp/ccatZFLR.s:986: Error: cannot honor width suffix -- `mov r0,#6'
/tmp/ccatZFLR.s:987: Error: cannot honor width suffix -- `neg r0,r0'

[mbed] ERROR: "/home/giacomo/software/workspaces/virtualenv/mbed/bin/python" returned error code 1.
[mbed] ERROR: Command "/home/giacomo/software/workspaces/virtualenv/mbed/bin/python -u /home/giacomo/software/project/smartcooler/test-mbed5/test/mbed-os-example-blinky/mbed-os/tools/make.py -t GCC_ARM -m NUCLEO_L073RZ --source . --build ./BUILD/NUCLEO_L073RZ/GCC_ARM -c -v -j 1" in "/home/giacomo/software/project/smartcooler/test-mbed5/test/mbed-os-example-blinky"
```
Other _things_ already tried:

1. Repeat the steps several time (always same results)
2. Try it without virtualenv (same)
3. Online compiler (Works with same target)
4. Other targets, like NUCLEO_F401RE (compiled)

theotherjimmy, 2017-10-30:
Hi @GiacomoThror. Thanks for the detailed bug report. Could you tell us what version (or if on master what hash) of Mbed OS you are using? It looks like you are using a version >= 5.6.0 (you have a TARGET_CORTEX folder, which were introduced in 5.6.0). Also, could you tell us what version of `arm-non-eabi-gcc` you are using. to do that you can run the command `arm-none-eabi-gcc --version` which should give output similar to (taken from my arch machine with a very new gcc):
```
# arm-none-eabi-gcc --version
arm-none-eabi-gcc (Arch Repository) 7.2.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Using Mbed OS master and the above version of GCC, and a release from the 6.x series, I cannot reproduce the problem.

GiacomoThror, 2017-11-04:
Hi @theotherjimmy,

I solved the problem! 
As you suggested the problem is the version of `arm-non-eabi-gcc`. I have the distribution old package version 4.9.3. Now installed the suggested 6.3.1 and it works.
For the mbed-os repository I had tried with both branch "master" and "mbed-os-5.6" with same results, the issue was not here.

Thanks a lot.

closed 2017-11-04 by GiacomoThror

theotherjimmy, 2017-11-06:
Thank you for posting how you solved your problem. I hope this may help others in the future.