ARMmbed/mbed-cli #557: mbed targets --supported is broken

BlackstoneEngineering, 2017-10-31:
TLDR:
the command `mbed target --supported` fails, but `mbed compiler --supported` and `mbed compiler -S` works. This command documented in videos and tutorials and needs to work.

Long Version:
On mbed CLI v1.2.0 the `mbed target --supported` command works, but on v1.2.2 it does not. Something changed here that broke the command. 

The output failure message is `[mbed] ERROR: Unknown Error: sequence item 4: expected string, bool found`, which looks like a python issue, but not sure. 

This was found on windows 10, using cmd. 

```shell
C:\Users\ausbla01\mbed\mbed-os>mbed target --supported
[mbed] WARNING: Could not find mbed program in current path "C:\Users\ausbla01\mbed\mbed-os".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
+----------------------+-----------+-----------+-----------+-----------+-----------+-----------+
| Target               | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |   ARMC6   |
+----------------------+-----------+-----------+-----------+-----------+-----------+-----------+
| ARCH_PRO             | Supported | Supported | Supported | Supported | Supported | Supported |
| ARM_BEETLE_SOC       | Supported | Supported | Supported | Supported | Supported | Supported |
| ARM_CM3DS_MPS2       | Supported | Supported | Supported | Supported | Supported | Supported |
| B96B_F446VE          | Supported | Supported | Supported | Supported | Supported | Supported |
| DELTA_DFBM_NQ620     | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_F413ZH         | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_F429ZI         | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_F469NI         | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_F746NG         | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_F769NI         | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_L072CZ_LRWAN1  | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_L475VG_IOT01A  | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_L476VG         | Supported | Supported | Supported | Supported | Supported | Supported |
| EFM32GG_STK3700      | Supported | Supported | Supported | Supported | Supported | Supported |
| EFM32LG_STK3600      | Supported | Supported | Supported | Supported | Supported | Supported |
| EFM32PG12_STK3402    | Supported | Supported | Supported | Supported | Supported | Supported |
| EFM32PG_STK3401      | Supported | Supported | Supported | Supported | Supported | Supported |
| EFM32WG_STK3800      | Supported | Supported | Supported | Supported | Supported | Supported |
| FF1705_L151CC        |     -     | Supported | Supported | Supported | Supported | Supported |
| HEXIWEAR             | Supported | Supported | Supported | Supported | Supported | Supported |
| K22F                 | Supported | Supported | Supported | Supported | Supported | Supported |
| K64F                 | Supported | Supported | Supported | Supported | Supported | Supported |
| K66F                 | Supported | Supported | Supported | Supported | Supported | Supported |
| K82F                 | Supported | Supported | Supported | Supported | Supported | Supported |
| KL25Z                | Supported | Supported | Supported | Supported | Supported | Supported |
| KL43Z                | Supported | Supported | Supported | Supported | Supported | Supported |
| KL46Z                | Supported | Supported | Supported | Supported | Supported | Supported |
| KL82Z                | Supported | Supported | Supported | Supported | Supported | Supported |
| KW24D                | Supported | Supported | Supported | Supported | Supported | Supported |
| KW41Z                | Supported | Supported | Supported | Supported | Supported | Supported |
| LPC1768              | Supported | Supported | Supported | Supported | Supported | Supported |
| LPC1769              | Supported | Supported | Supported | Supported | Supported | Supported |
| LPC4088              | Supported | Supported | Supported | Supported | Supported | Supported |
| LPC4088_DM           | Supported | Supported | Supported | Supported | Supported | Supported |
| LPC54114             | Supported | Supported | Supported | Supported | Supported | Supported |
| LPC546XX             | Supported | Supported | Supported | Supported | Supported | Supported |
| MAX32600MBED         | Supported | Supported | Supported | Supported | Supported | Supported |
| MAX32620HSP          | Supported | Supported | Supported | Supported | Supported | Supported |
| MAX32625MBED         | Supported | Supported | Supported | Supported | Supported | Supported |
| MAX32625NEXPAQ       | Supported | Supported | Supported | Supported | Supported | Supported |
| MAX32630FTHR         | Supported | Supported | Supported | Supported | Supported | Supported |
| MAXWSNENV            | Supported | Supported | Supported | Supported | Supported | Supported |
| MTS_DRAGONFLY_F411RE | Supported | Supported | Supported | Supported | Supported | Supported |
| MTS_MDOT_F411RE      | Supported | Supported | Supported | Supported | Supported | Supported |
| NCS36510             | Supported | Supported | Supported | Supported | Supported | Supported |
| NRF51_DK             | Supported | Supported | Supported | Supported | Supported | Supported |
| NRF51_DONGLE         | Supported | Supported | Supported | Supported | Supported | Supported |
| NRF52840_DK          | Supported | Supported | Supported | Supported | Supported | Supported |
| NRF52_DK             | Supported | Supported | Supported | Supported | Supported | Supported |
| NUCLEO_F070RB        | Supported | Supported | Supported | Supported | Supported | Supported |
| NUCLEO_F072RB        | Supported | Supported | Supported | Supported | Supported | Supported |
| NUCLEO_F091RC        | Supported | Supported | Supported | Supported | Supported | Supported |
[... 178 lines cut ...]
| NUMAKER_PFM_NUC472   |     -     | Supported | Supported | Supported | Supported | Supported |
| REALTEK_RTL8195AM    |     -     | Supported | Supported | Supported | Supported | Supported |
| RO359B               | Supported | Supported | Supported | Supported | Supported | Supported |
| SARA_NBIOT_EVK       |     -     | Supported | Supported | Supported | Supported | Supported |
| SILICA_SENSOR_NODE   |     -     | Supported | Supported | Supported | Supported | Supported |
| TB_SENSE_1           | Supported | Supported | Supported | Supported | Supported | Supported |
| TB_SENSE_12          | Supported | Supported | Supported | Supported | Supported | Supported |
| TMPM066              |     -     | Supported | Supported | Supported | Supported | Supported |
| TY51822R3            | Supported | Supported | Supported | Supported | Supported | Supported |
| UBLOX_C027           | Supported | Supported | Supported | Supported | Supported | Supported |
| UBLOX_C030_N211      |     -     | Supported | Supported | Supported | Supported | Supported |
| UBLOX_C030_U201      |     -     | Supported | Supported | Supported | Supported | Supported |
| UBLOX_EVA_NINA       | Supported | Supported | Supported | Supported | Supported | Supported |
| UBLOX_EVK_NINA_B1    | Supported | Supported | Supported | Supported | Supported | Supported |
| UBLOX_EVK_ODIN_W2    |     -     | Supported | Supported | Supported | Supported | Supported |
| UBRIDGE              | Supported | Supported | Supported | Supported | Supported | Supported |
| USENSE               | Supported | Supported | Supported | Supported | Supported | Supported |
| WIZWIKI_W7500        | Supported | Supported | Supported | Supported | Supported | Supported |
| WIZWIKI_W7500ECO     | Supported | Supported | Supported | Supported | Supported | Supported |
| WIZWIKI_W7500P       | Supported | Supported | Supported | Supported | Supported | Supported |
| XDOT_L151CC          |     -     | Supported | Supported | Supported | Supported | Supported |
+----------------------+-----------+-----------+-----------+-----------+-----------+-----------+
Supported targets: 96

```

theotherjimmy, 2017-10-31:
Thanks for the bug report. I may have time to look at this bug this week. maybe not though.

andcor02, 2017-11-13:
I am also getting the same issue

BlackstoneEngineering, 2017-11-28:
Latest cli, latest mbed-os, windows 10. 

```
C:\Users\ausbla01\mbed\mbed-os>mbed target --supported
[mbed] WARNING: Could not find mbed program in current path "C:\Users\ausbla01\mbed\mbed-os".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
[mbed] ERROR: Unknown Error: sequence item 4: expected string, bool found
---

C:\Users\ausbla01\mbed\mbed-os>mbed toolchain --supported
[mbed] WARNING: Could not find mbed program in current path "C:\Users\ausbla01\mbed\mbed-os".
[mbed] WARNING: You can fix this by calling "mbed new ." in the root of your program.
---
[mbed] ERROR: Unknown Error: sequence item 4: expected string, bool found
---

C:\Users\ausbla01\mbed\mbed-os>git log -n1
commit 70e7b40468854d33431889a9cd415364c00a2501
Merge: 33c972666 ff6bc08c5
Author: Jimmy Brisson <theotherjimmy@gmail.com>
Date:   Mon Nov 27 10:37:40 2017 -0600

    Merge pull request #5583 from kjbracey-arm/delete_null

    Remove pointless null check in operator delete

C:\Users\ausbla01\mbed\mbed-os>mbed --version
1.2.2
```

screamerbg, 2018-03-23:
This is now fixed. Thanks for reporting this

closed 2018-03-23 by screamerbg