ARMmbed/mbed-cli #558: mbed test error

RozenAstrayChen, 2017-11-01:
Hello i got this message on my __OSX__
```shell==
chenjiaqide-MacBook-Air:MQTT_IDW01 Rozen_mac$ mbed test --compile-list
Traceback (most recent call last):
  File "/Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/.temp/tools/test.py", line 30, in <module>
    from tools.test_api import test_path_to_name, find_tests, get_test_config, print_tests, build_tests, test_spec_from_test_builds
  File "/Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/.temp/tools/test_api.py", line 53, in <module>
    import tools.test_configs as TestConfig
ImportError: No module named test_configs
[mbed] ERROR: "/Users/Rozen_mac/anaconda/bin/python" returned error code 1.
[mbed] ERROR: Command "/Users/Rozen_mac/anaconda/bin/python -u /Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/.temp/tools/test.py --list -t GCC_ARM -m NUCLEO_F401RE --source /Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01" in "/Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01"
---
chenjiaqide-MacBook-Air:MQTT_IDW01 Rozen_mac$ mbed test -m K64F -t GCC_ARM
Traceback (most recent call last):
  File "/Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/.temp/tools/test.py", line 30, in <module>
    from tools.test_api import test_path_to_name, find_tests, get_test_config, print_tests, build_tests, test_spec_from_test_builds
  File "/Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/.temp/tools/test_api.py", line 53, in <module>
    import tools.test_configs as TestConfig
ImportError: No module named test_configs
[mbed] ERROR: "/Users/Rozen_mac/anaconda/bin/python" returned error code 1.
[mbed] ERROR: Command "/Users/Rozen_mac/anaconda/bin/python -u /Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/.temp/tools/test.py -t GCC_ARM -m K64F --source /Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01 --build /Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/BUILD/tests/K64F/GCC_ARM --test-spec /Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01/BUILD/tests/K64F/GCC_ARM/test_spec.json" in "/Users/Rozen_mac/code/fucking_sensor/MQTT_IDW01"
```
look like is 
```shell==
import tools.test_configs as TestConfig
ImportError: No module named test_configs
```

RozenAstrayChen, 2017-11-01:
I found my question , I don't know why I lost ```test_configs``` floder

theotherjimmy, 2017-11-01:
@RozenAstrayChen Thanks for telling us what went wrong. I'm going to close this issue as resolved. Please re open if you want more help.

closed 2017-11-01 by theotherjimmy