ARMmbed/mbed-cli #559: question about "--no-archive"

tz-arm, 2017-11-08:
Seems there is no explanation and no help comments for the parameter of "--no-archive"
Is it relevant to gcc "--no-whole-archive"?

theotherjimmy, 2017-11-08:
> Is it relevant to gcc "--no-whole-archive"?

I don't think so

That option is used in conjunction with `--library` to build a bunch of C/C++/ASM files as a static library. The `--no-archive` flag suppresses the creation of the `.a` file, and the deletion of the intermediate `.o` files.

theotherjimmy, 2017-11-28:
Marked as documentation, as there is no explanation of the command line option mentioned in this issue.

screamerbg, 2018-05-26:
@AnotherButler Could you please discuss with @theotherjimmy what's the right explanation/terminology for the `--no-archive` option?

theotherjimmy, 2018-05-27:
@screamerbg @AnotherButler I think I did this a while back: https://github.com/ARMmbed/Handbook/pull/394

@tz-arm Does the current documentation answer your question?

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2018-10-23 by screamerbg