ARMmbed/mbed-cli #560: Speed up `mbed-cli new` command by performing shallow clone of mbed-os

cmonr, 2017-11-09:
When running `mbed-cli new <name>`, the entire repo history of mbed-os is cloned into the user's repository.

At the current size of the mbed-os repo, the `git clone <mbed-os repo>` command takes ~3 minutes on my system.
Replacing this with `git clone -b latest --depth=1 <mbed-os repo>` cuts this time down to ~1.5 minutes.

teetak01, 2017-11-10:
Maybe instead of being an default option, there could be an optional configurational parameter to deploy Mbed OS as shallow. This way it would affect equally `import`, `deploy `and `new`.

bridadan, 2017-11-10:
@cmonr You can actually already do this with the option `--depth [DEPTH]`. It's currently present in `mbed import`, `mbed add`, `mbed deploy`, and `mbed update`.

I don't think we can make this the default option like @teetak01 mentioned because the `.lib` files that are then turned into repositories can reference any commit in the repository's history. If the depth is set too shallow, this commit won't be present and it will error.

cmonr, 2017-11-10:
@bridadan Using `--depth` on its own is definitely not enough and can cause the entire clone to fail, but (from what I can tell) all that goes into `mbed-os.lib` is the git URL with the specific commit hash. You can still get the same hash if `-b latest` is used, with the benefit of a faster clone and less disk space used.

bridadan, 2017-11-10:
Ah I see, so you're saying just for the `mbed new` command (which always clones from the `latest` tag), just run `git clone https://github.com/ARMmbed/mbed-os -b latest` since that will be faster?

Seems fine to me, do you need some pointers on where to make the change? This is the relevant section of the CLI: https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L1727. You'll probably need to look into the `Git` abstractions as well: https://github.com/ARMmbed/mbed-cli/blob/master/mbed/mbed.py#L610

cmonr, 2017-11-16:
Yes, and no. More specifically, right now the `mbed new` command does a clone of the entire repo, _then_ switches to the `latest` branch.

Both `--depth=1` and `-b latest` are needed, the former to speed up the clone, and the latter to make sure that all files needed are actually pulled.

bridadan, 2017-11-17:
Ah, I see, thanks for the explanation. I think that should be fine, but I think the final call may end up being @screamerbg's.

@theotherjimmy Do you know if this will affect the online compiler? Typically you can always go back in commits in the online compiler. If the online compiler uses mbed CLI in the back end then this would be a breaking change.

theotherjimmy, 2017-11-20:
The online compiler does not use Mbed CLI, as it's directory structure is very very different.

screamerbg, 2018-05-26:
@cmonr With the `mbed cache` feature now being default, do you still think we need this?

cmonr, 2018-05-26:
OH. Forgot this was still open.

This can be closed. The crux of my problem was Shared Directory performance in VirtualBox.
But I agree that now that caching is a thing, things should be much faster.

closed 2018-05-26 by cmonr