ARMmbed/mbed-cli #561: Mbed CLI should do shallow clone by default on import

janjongboom, 2017-11-11:
On slow connections (and on some fast ones too), importing an Mbed OS program using CLI is extremely slow and might take up to half an hour. Enabling cache helps, but not the first time - so bad out of box experience -, plus users forget it.

This is mainly due to the time it costs to clone mbed-os. On the super-fast Seoul internet here (110 mbps down) it takes **3 minutes, 21 seconds** to clone mbed-os. However, 99% of the users do not need the full history, they just want to build and are probably not going to contribute back to the project. So we could speed this up significantly if we'd shallow clone the repo when importing. So I'd suggest to use `--depth=1` (I see it's already in CLI) by default.

Shallow import on the same internet takes me 36 seconds.

cc @toyowata @neil-tan

janjongboom, 2017-11-11:
Ah, dupe of #560.

theotherjimmy, 2017-11-28:
Closing this issue in favor of #560

closed 2017-11-28 by theotherjimmy