ARMmbed/mbed-cli #562: Progress bar when importing

janjongboom, 2017-11-11:
Could we show some progress while importing? Maybe just normal git progress?

Currently we give no clue to a user how far the import process is.

theotherjimmy, 2017-11-13:
Marked as "help wanted" because I'm bogged down in other things right now.

MarceloSalazar, 2017-11-13:
Feature already requested: https://github.com/ARMmbed/mbed-cli/issues/136

bridadan, 2017-11-13:
@janjongboom Should be straightforward enough, but like Jimmy said I'm loaded with other things at the moment.

If you run `mbed import` with the `-v` flag it will output the `git` command output, which includes the progress.

If that isn't good enough, we could look into maybe parsing the git output as its printed and just getting the progress info.

screamerbg, 2017-12-22:
@janjongboom @bridadan @MarceloSalazar PR #593 introduces this.

MarceloSalazar, 2017-12-23:
@screamerbg looks good, thanks!

janjongboom, 2017-12-29:
@screamerbg Nice!

screamerbg, 2018-02-06:
Now available in Mbed CLI 1.4.0

closed 2018-02-06 by screamerbg