ARMmbed/mbed-cli #564: Traceback error when 'mbed detect' on mercurial project

MarceloSalazar, 2017-11-16:
Error when using `mbed detect` on mercurial repository.
Instructions to reproduce:

``
$ mbed import https://os.mbed.com/teams/mbed/code/mbed_blinky/
$ cd mbed_blinky
$ mbed detect
Traceback (most recent call last):
  File "/Users/.../mbed_blinky/.temp/tools/detect_targets.py", line 33, in <module>
    from tools.test_api import get_autodetected_MUTS_list
  File "/Users/.../mbed_blinky/.temp/tools/test_api.py", line 53, in <module>
    import tools.test_configs as TestConfig
ImportError: No module named test_configs
[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /Users/.../mbed_blinky/.temp/tools/detect_targets.py" in "/Users/.../mbed_blinky"
```

`mbedls` works fine, so I assume the problem is in `mbed detect`

mbed-cli 1.2.2
python 2.7.14
OS X El Capitan

@theotherjimmy

theotherjimmy, 2017-11-16:
Thanks. I'll include `test_configs` in the Mbed OS 2 tools release I will be making to accompany the Mbed OS 5.6.5 release.

MarceloSalazar, 2017-11-16:
Interestingly, I'm also reproducing this on a Mbed OS 5 project.

theotherjimmy, 2017-11-16:
Do you have that traceback?

MarceloSalazar, 2017-11-16:
I take that back; my Mbed OS 5 project was being detected as OS 2, but now works ok.
The issue is still valid for a Mbed OS 2 project.

screamerbg, 2017-12-02:
@theotherjimmy Bump. This is related to mbed 2 SDK tools.

screamerbg, 2017-12-21:
Now fixed in mbed CLI 1.3 release

closed 2017-12-21 by screamerbg