ARMmbed/mbed-cli #566: mbed test

phobos3, 2017-11-27:
1)  Please add to the documentation that new green tea test main.cpp files must be two folders deep in the TESTS directory.  This is not mentioned anywhere and took forever to discover!

2)  Your example seems to imply you can have a file with main {} in the root of an mbed program, but when I try to run mbed test, I get a compiler error.  Did I do something wrong or can you not really run mbed test in the root of your program ( b/c you need your program's main to live somewhere, right?)

theotherjimmy, 2017-11-28:
Thanks for the feedback!

Re 1) Where would you expect to find that information?

Re 2) You did nothing wrong; You can not run mbed test from the root of a program. You are correct that the program's main will conflict with the test main.

phobos3, 2017-11-28:
Thank you for the quick response!!!

1)  Section 8 of the README.md for starters, and what really through me off is that in the greentea section of the mbed handbook it says """
Host test detection

When developing with mbed OS, Greentea detects host tests automatically if you place them in the correct location. All tests in mbed OS are placed under a TESTS directory. You may place custom host test scripts in a folder named host_tests in this folder. For more information about the mbed OS test directory structure, please see the mbed CLI documentation."

Maybe add a note there as well.

2)  Well then I guess I would request a feature add where the tool can ignore specifically named .cpp files, like maybe "app-main.cpp" or something.  Greentea is great, and we're really excited to get it into our CI flow!  But as it stands, I'll probably have to wrap the call to mbed test with a little script to temporarily rename our program main to a different file extension.

theotherjimmy, 2017-11-28:
I opened a handbook PR, armmbed/handbook#343, which I hope improves upon the text you just pointed out. [Rendered](https://github.com/ARMmbed/Handbook/pull/343/files?short_path=745d2f2)

screamerbg, 2018-03-23:
@Alex-EEE Are you happy to close the ticket if resolved?

closed 2018-03-23 by screamerbg

Alex-EEE, 2018-04-03:
Sorry I didn't close this!  Yes, looks good, thanks for asking!