ARMmbed/mbed-cli #567: test found but won't run

phobos3, 2017-11-28:
Hi,

I've added a custom tests to my program root.  It can find it with "mbed test --compile-list" like so:
...
Test Case:
    Name: mbed-os-rtos-target_cortex-rtx5-tests-memory-heap_and_stack
    Path: .\mbed-os\rtos\TARGET_CORTEX\rtx5\TESTS\memory\heap_and_stack
Test Case:
    Name: tests-alex-alex-sub
    Path: .\TESTS\alex\alex-sub

However when I try to run it, it says "mbedgt: no platform/target matching tests were found!"  Do I need to add any information to my test folder / .cpp file to tie it to a particular platform?

Full output:
"""
mbedgt: greentea test automation tool ver. 1.3.2
mbedgt: test specification file 'C:\firmware\mbed\mbed-os-example-blinky\BUILD\tests\k64f\IAR\test_spec.json' (specified
 with --test-spec option)
mbedgt: using 'C:\firmware\mbed\mbed-os-example-blinky\BUILD\tests\k64f\IAR\test_spec.json' from current directory!
("TestSpec::load('C:\\firmware\\mbed\\mbed-os-example-blinky\\BUILD\\tests\\k64f\\IAR\\test_spec.json')", "[Errno 2] No
such file or directory: 'C:\\\\firmware\\\\mbed\\\\mbed-os-example-blinky\\\\BUILD\\\\tests\\\\k64f\\\\IAR\\\\test_spec.
json'")
mbedgt: detecting connected mbed-enabled devices...
[1511887239.27][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_
mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1511887239.37][root]Found htm target id, 02400201CCA63E533158C3EB, for usb target id 02400201CCA63E533158C3EB
mbedgt: detected 1 device
mbedgt: shuffle seed: 0.1660256276
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!
mbedgt: completed in 0.17 sec
mbedgt: exited with code -110
[mbed] ERROR: "mbedgt" returned error code -110.

theotherjimmy, 2017-11-28:
@phobos3 That error is trying to inform you that your platform does not match the test that you compiled.

phobos3, 2017-11-28:
"your platform does not match the test that you compiled."  How does it match tests with platforms?  I'm just trying to run the Greentea example test, attached below.  

Do you mean the K64 folder inside of BUILD/tests ?

When I run the mbed-os tests, they compile and run as I would expect.  What I'm trying right now, is disabling the mbed-os tests by renaming the TESTS folder to "skip", and just trying to run my own test and see a pass result.  When I re enable those tests, they start running, so I think everything is setup correctly on the back end:

mbedgt: greentea test automation tool ver. 1.3.2
mbedgt: test specification file 'C:\firmware\mbed\mbed-os-example-blinky\BUILD\tests\k64f\IAR\test_spec.json' (specified
 with --test-spec option)
mbedgt: using 'C:\firmware\mbed\mbed-os-example-blinky\BUILD\tests\k64f\IAR\test_spec.json' from current directory!
mbedgt: detecting connected mbed-enabled devices...
[1511889863.33][mbedls.lstools_base]Call to deprecated function list_mbeds_ext. Functionality has been moved into 'list_
mbeds'. Please use list_mbeds with 'unique_names=True' and 'read_details_txt=True'
[1511889863.35][root]Found htm target id, 02400201CCA63E533158C3EB, for usb target id 02400201CCA63E533158C3EB
mbedgt: detected 1 device
mbedgt: processing target 'K64F' toolchain 'IAR' compatible platforms... (note: switch set to --parallel 1)
mbedgt: running 70 tests for platform 'K64F' and toolchain 'IAR'
mbedgt: mbed-host-test-runner: started

*******

Test Im trying to add:

using namespace utest::v1;

// A test that returns successfully is considered successful
void test_success() {
    TEST_ASSERT(true);
}

// Tests that assert are considered failing
void test_failure() {
    TEST_ASSERT(false);
}

utest::v1::status_t test_setup(const size_t number_of_cases) {
    // Setup Greentea using a reasonable timeout in seconds
    GREENTEA_SETUP(40, "default_auto");
    return verbose_test_setup_handler(number_of_cases);
}

// Test cases
Case cases[] = {
    Case("Testing success test", test_success),
    Case("Testing failure test", test_failure),
};

Specification specification(test_setup, cases);

// Entry point into the tests
int main() {
    return !Harness::run(specification);
}

theotherjimmy, 2017-11-28:
Maybe your test is not being found? could you try `mbed test -l -m K64F -t IAR` with your test?

phobos3, 2017-11-28:
Result.  Looks good to me, that last test "alex-sub" is what I'm trying to run.

PS C:\firmware\mbed\mbed-os-example-blinky> mbed test -l -m K64F -t IAR
Test Case:
    Name: mbed-os-features-frameworks-utest-tests-unit_tests-basic_test
    Path: .\mbed-os\features\frameworks\utest\TESTS\unit_tests\basic_test
...
Test Case:
    Name: mbed-os-rtos-target_cortex-rtx5-tests-memory-heap_and_stack
    Path: .\mbed-os\rtos\TARGET_CORTEX\rtx5\TESTS\memory\heap_and_stack
Test Case:
    Name: tests-alex-alex-sub
    Path: .\TESTS\alex\alex-sub
mbedgt: greentea test automation tool ver. 1.3.2
mbedgt: test specification file 'C:\firmware\mbed\mbed-os-example-blinky\BUILD\tests\K64F\IAR\test_spec.json' (specified
 with --test-spec option)
mbedgt: using 'C:\firmware\mbed\mbed-os-example-blinky\BUILD\tests\K64F\IAR\test_spec.json' from current directory!
("TestSpec::load('C:\\firmware\\mbed\\mbed-os-example-blinky\\BUILD\\tests\\K64F\\IAR\\test_spec.json')", "[Errno 2] No
such file or directory: 'C:\\\\firmware\\\\mbed\\\\mbed-os-example-blinky\\\\BUILD\\\\tests\\\\K64F\\\\IAR\\\\test_spec.
json'")
PS C:\firmware\mbed\mbed-os-example-blinky>

phobos3, 2017-11-28:
Alternatively, can you just tell me what you needed to do to get this file to run as a test under whatever mbed program you happen to have lying around?  (I'm testing with blinky right now)

[main (2).zip](https://github.com/ARMmbed/mbed-cli/files/1511156/main.2.zip)

theotherjimmy, 2017-11-28:
I think we discussed that you cannot have a test and an application in the same project over in #566

phobos3, 2017-11-28:
Correct, I understand I can't have a main.cpp outside of test, but what I'm trying to add a custom test into a collection of mbed modules, similar to what's in the mbed-cli README

mbed-os-program
 |- pqr.lib             # Required libs
 |- xyz.lib
 |- test-app   # new module
 | |- TESTS
 | | |  alex
 | | | | alex-sub
 | | | | | main.cpp   # example greentea test file
 |- mbed-os
 |  |- frameworks        # Test dependencies
 |  |  `_greentea-client # Greentea client required by tests.
 |  |...
 |  `- TESTS              # Tests directory. Special name upper case TESTS is excluded during application build process
 |     |- TestGroup1      # Test Group directory
 |     |  `- TestCase1    # Test case source directory
 |     |      `- main.cpp # Test source
 |     |- TestGroup2
 |     |   `- TestCase2
 |     |      `- main.cpp
 |     `- host_tests      # Python host tests script directory
 |        |- host_test1.py
 |        `- host_test2.py
 `- build                 # Build directory
     |- <TARGET>          # Target directory
     | `- <TOOLCHAIN>     # Toolchain directory
     |   |- TestCase1.bin # Test binary
     |   `- TestCase2.bin
     | ....

theotherjimmy, 2017-11-28:
Using the following dir structure:
```
.
├── TESTS
│  └── alex
│     └── alex_sub
│        └── main.cpp
```
Worked with a K64F mounted and `mbed test -t gcc_arm -m k64F`

theotherjimmy, 2017-11-28:
I also just tried with:
```
TESTS
└── alex
   └── alex-sub
      └── main.cpp
```
and that worked too.

bridadan, 2017-11-28:
Hi @phobos3, not sure if I understand your whole problem, but I noticed the following line in your comments:

> What I'm trying right now, is disabling the mbed-os tests by **renaming the TESTS folder to "skip"**, and just trying to run my own test and see a pass result. 

Renaming the `TESTS` folder to `skip` will not make them be skipped, **instead it will cause each of the tests to be included in the "common code"**. If you have added other tests, then this will almost certainly result in a link/compile error (specifically, multiple `main` functions).

You can either remove that folder or place an `.mbedignore` file with a contents of `*` just inside the TESTS directory:

```
- /TESTS
    - /.mbedignore (contents should just be "*" with no quotes)
```

If you're still having issues, be sure your tests are actually building correctly (you may need to scroll up in the log to see the build errors).

phobos3, 2017-11-28:
@bridadan That did the trick!  I wasn't quite understanding the messages I was getting from the tool, but now I can see that it was trying to build the test files into the library and it was failing, but the script continued to execute, but found no files to load b/c the build had been terminated!
*****
For anyone reading this later, if you see this, you probably have the same issue:

[Fatal Error] main.cpp@28,0: [Pe035]: #error directive: [NOT_SUPPORTED] test not supported
[DEBUG] Return: 2
[DEBUG] Output:
[DEBUG] Output:     #error [NOT_SUPPORTED] test not supported
[DEBUG] Output:      ^
[DEBUG] Output: "C:\firmware\mbed\mbed-os-example-blinky\mbed-os\skip\mbed_drivers\mem_trace\main.cpp",28  Fatal error[P
e035]: #error directive: [NOT_SUPPORTED] test not supported
[DEBUG] Output: Fatal error detected, aborting.
Failed to build library

Memory map breakdown for built projects (values in Bytes):
+------+--------+-----------+------------+-------------+
| name | target | toolchain | static_ram | total_flash |
+------+--------+-----------+------------+-------------+
+------+--------+-----------+------------+-------------+

theotherjimmy, 2017-12-01:
It sounds like the problem has been resolved. Closing.

closed 2017-12-01 by theotherjimmy