ARMmbed/mbed-cli #568: Replace SHAs withgit refs in .lib files

cmonr, 2017-11-28:
This is related to PR https://github.com/ARMmbed/mbed-cli/pull/565, but not dependent on it.

This is an enhancement that:
1. should help clarify exactly what versions/releases all .lib repos are at, and
2. allows for shallow clones in https://github.com/ARMmbed/mbed-cli/pull/565 to function, since the SHA must be tied to a git ref.

cmonr, 2017-11-28:
Actually, branches could be used as well (specifically, branch heads).

If the reference is listed under `git ls-remote`, then it can be directly fetched from the remote, either from the ref name or the SHA.

theotherjimmy, 2017-11-28:
@cmonr We have a policy of not allowing branches in `.lib` and `.bld` files, as they _can move_. We have defined library references to be fixed, and updates require _explicit_ user requests. As such, I can't accept a PR that automatically swaps a SHA ref; slow clone, can't change meaning; to a branch ref; fast clone, can change meaning.

closed 2018-05-26 by screamerbg