ARMmbed/mbed-cli #569: Mbed CLI can leak GitHub OAuth tokens

janjongboom, 2017-11-29:
See https://github.com/johanstokking/arm-ttn-webinar/blob/f6f9cb9871d91ee664c8749fe8e0ee0a3e12e978/mbed-os.lib

For some reason there is a *valid* OAuth token leaked into mbed-os.lib. Any idea what this can be, and how we can prevent this?! This is a very serious security issue which could potentially lead to exposing our user's full GH accounts.

cc @johanstokking

janjongboom, 2017-11-29:
We should strip off all authentication parts in the URL before filling in the .lib file.

johanstokking, 2017-11-29:
Reproduce:

```
$ mbed new .
[mbed] Creating new program "test-app" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch latest
[mbed] Updating reference "mbed-os" -> "https://SECRET:x-oauth-basic@github.com/ARMmbed/mbed-os/#182bbd51bc8d1b6217477c2b5de3f1d0c5e6249f"
```

This is in my `~/.gitconfig`:
```
$ cat .gitconfig
# This is Git's per-user configuration file.
...
[url "https://SECRET:x-oauth-basic@github.com/"]
	insteadOf = https://github.com/
...
```

Why? Because [some tools need to get private stuff over https](https://gist.github.com/shurcooL/6927554).

This should _never_ be placed as-is in a repository by any tool.

In this particular case, Github was kind enough to automatically revoke this personal access token immediately.

theotherjimmy, 2017-11-29:
@janjongboom @johanstokking Thank you very much for this bug report! I will look into resolving this today.

screamerbg, 2017-12-21:
Now fixed in mbed CLI 1.3 release

closed 2017-12-21 by screamerbg