ARMmbed/mbed-cli #575: Mbed CLI doesn't return supported targets with pre-Mbed OS 5.5

screamerbg, 2017-12-02:
Broken backwards compatibility. Any developer using previous versions of Mbed OS cannot list supported targets if they update Mbed CLI to v1.2+.

Suspected that this was introduced with https://github.com/ARMmbed/mbed-cli/pull/460

CC @theotherjimmy 

```
$ mbed compile --supported
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS] [-S]
               [-f GENERAL_FILTER_REGEX] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rtos] [--rpc] [--eth]
               [--usb_host] [--usb] [--dsp] [--fat] [--ublox] [--testlib]
               [-l LINKER_SCRIPT]
make.py: error: unrecognized arguments: matrix
[mbed] ERROR: "/usr/bin/python" returned error code 2.
```

cmonr, 2017-12-03:
@screamerbg What's your `mbed --version` output? On 1.2.2, `mbed compile --supported` works for me.

Also, PR https://github.com/ARMmbed/mbed-cli/pull/572 happens to address issues that came up with #460.

screamerbg, 2017-12-03:
@cmonr Changed the title so it's clearer.

Simple way to reproduce this:

```
/tmp/1:$ mbed --version
1.2.2
/tmp/1:$ mbed new test-program
[mbed] Creating new program "test-program" (git)
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch latest
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#182bbd51bc8d1b6217477c2b5de3f1d0c5e6249f"

/tmp/1:$ cd test-program/
/tmp/1/test-program:$ mbed compile --supported
+----------------------+-----------+-----------+-----------+-----------+-----------+-----------+
| Target               | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |   ARMC6   |
+----------------------+-----------+-----------+-----------+-----------+-----------+-----------+
| ARCH_PRO             | Supported | Supported | Supported | Supported | Supported | Supported |
| ARM_BEETLE_SOC       | Supported | Supported | Supported | Supported | Supported | Supported |
| ARM_CM3DS_MPS2       | Supported | Supported | Supported | Supported | Supported | Supported |
| B96B_F446VE          | Supported | Supported | Supported | Supported | Supported | Supported |
| DELTA_DFBM_NQ620     | Supported | Supported | Supported | Supported | Supported | Supported |
| DISCO_F413ZH         | Supported | Supported | Supported | Supported | Supported | Supported |

[snip]

Supported targets: 103

/tmp/1/test-program:$ cd mbed-os
/tmp/1/test-program/mbed-os:$ mbed update mbed-os-5.3.1
[mbed] Updating library "mbed-os" to branch mbed-os-5.3.1
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#0789928ee7f2db08a419fa4a032fffd9bd477aa7"

/tmp/1/test-program/mbed-os:$ mbed update mbed-os-5.4.1
[mbed] Updating library "mbed-os" to branch mbed-os-5.4.1
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#f4864dc6429e1ff5474111d4e0f6bee36a759b1c"

/tmp/1/test-program/mbed-os:$ mbed compile --supported
usage: make.py [-h] [-m MCU] [-t TOOLCHAIN] [--color] [--cflags CFLAGS]
               [--asmflags ASMFLAGS] [--ldflags LDFLAGS] [-c]
               [--profile PROFILE] [--app-config APP_CONFIG] [-p PROGRAM]
               [-n PROGRAM] [-j JOBS] [-v] [--silent] [-D MACROS] [-S]
               [-f GENERAL_FILTER_REGEX] [--automated] [--host HOST_TEST]
               [--extra EXTRA] [--peripherals PERIPHERALS]
               [--dep DEPENDENCIES] [--source SOURCE_DIR]
               [--duration DURATION] [--build BUILD_DIR] [-N ARTIFACT_NAME]
               [-d DISK] [-s SERIAL] [-b BAUD] [-L] [--rtos] [--rpc] [--eth]
               [--usb_host] [--usb] [--dsp] [--ublox] [--testlib]
               [-l LINKER_SCRIPT]
make.py: error: unrecognized arguments: matrix
[mbed] ERROR: "/usr/bin/python" returned error code 2.
[mbed] ERROR: Command "/usr/bin/python -u /private/tmp/1/test-program/mbed-os/tools/make.py -S matrix" in "/private/tmp/1/test-program/mbed-os"
```

screamerbg, 2017-12-21:
Now fixed with mbed CLI 1.3 release

closed 2017-12-21 by screamerbg