ARMmbed/mbed-cli #577: Let 'mbed add' use tag names

seankinghan, 2017-12-04:
mbed CLI currently supports tags for 'mbed update', but not for 'add'. It would be very useful to have this, both for the learning benefits of uniformity across the tool, and also for the same functionality benefits as are available for 'update'.

screamerbg, 2017-12-21:
@seankinghan mbed CLI supports tags/branches to both add and import, e.g. using the the syntax `mbed add/import <url>[#rev|branch|tag]`.

Here's an example:

```
$ mbed add https://github.com/ARMmbed/mbed-os#latest
[mbed] Adding library "mbed-os" from "https://github.com/ARMmbed/mbed-os" at branch latest
[mbed] Updating reference "mbed-os" -> "https://github.com/ARMmbed/mbed-os/#8b54959f5bca08a8ad44a701bafd634afbb38b03"

$ mbed releases -r
test (no revision)
  No release tags detected
`- mbed-os (#8b54959f5bca, tags: latest, mbed-os-5.7.0)
     * mbed-os-5.1.0 
     * mbed-os-5.1.1 
     * mbed-os-5.1.2 
     * mbed-os-5.1.3 
     * mbed-os-5.1.4 
     * mbed-os-5.1.5 
     * mbed-os-5.2.0 
     * mbed-os-5.2.1 
     * mbed-os-5.2.2 
     * mbed-os-5.2.3 
     * mbed-os-5.3.0 
     * mbed-os-5.3.1 
     * mbed-os-5.3.2 
     * mbed-os-5.3.3 
     * mbed-os-5.3.4 
     * mbed-os-5.3.5 
     * mbed-os-5.3.6 
     * mbed-os-5.4.0 
     * mbed-os-5.4.1 
     * mbed-os-5.4.2 
     * mbed-os-5.4.3 
     * mbed-os-5.4.4 
     * mbed-os-5.4.5 
     * mbed-os-5.4.6 
     * mbed-os-5.4.7 
     * mbed-os-5.5.0 
     * mbed-os-5.5.1 
     * mbed-os-5.5.2 
     * mbed-os-5.5.3 
     * mbed-os-5.5.4 
     * mbed-os-5.5.5 
     * mbed-os-5.5.6 
     * mbed-os-5.5.7 
     * mbed-os-5.6.0 
     * mbed-os-5.6.1 
     * mbed-os-5.6.2 
     * mbed-os-5.6.3 
     * mbed-os-5.6.4 
     * mbed-os-5.6.5 
     * mbed-os-5.6.6 
     * mbed-os-5.7.0  <- current
```

seankinghan, 2017-12-21:
Thanks. Can we have a look at the docs for it?

geoffrey-vl, 2018-03-05:
I've been trying to use this feature but it doesn't seem to be working for me at this stage.
Here is the output:

```
Microsoft Windows [Version 10.0.16299.248]
(c) 2017 Microsoft Corporation. Alle rechten voorbehouden.

C:\work-mbed\mbed-os-example-filesystem>
C:\work-mbed\mbed-os-example-filesystem>mbed remove dataflash-driver
[mbed] Removing library "dataflash-driver" in "C:\work-mbed\mbed-os-example-filesystem\dataflash-driver"

C:\work-mbed\mbed-os-example-filesystem>mbed add https://github.com/ARMmbed/dataflash-driver#g-fix-travis
[mbed] ERROR: Invalid revision (g-fix-travis)
---

C:\work-mbed\mbed-os-example-filesystem>mbed add -vv https://github.com/ARMmbed/dataflash-driver#g-fix-travis
[mbed] Working path "C:\work-mbed\mbed-os-example-filesystem" (program)
[mbed] Query "git remote -v" in C:\work-mbed\mbed-os-example-filesystem
origin  ssh://someserver/repos/delhi/mbed-os-example-filesystem (fetch)
origin  ssh://someserver/repos/delhi/mbed-os-example-filesystem (push)
[mbed] Query "git rev-parse HEAD" in C:\work-mbed\mbed-os-example-filesystem
699fe7e0986e87c0d3e9a7616ff7b241d31c188e
[mbed] ERROR: Invalid revision (g-fix-travis)
---

C:\work-mbed\mbed-os-example-filesystem>git --version
git version 2.14.1.windows.1
```

theotherjimmy, 2018-03-05:
Hi @geoffrey-vl,

You can only use tags and hashes in the URL to `mbed add`. That's probably related.

geoffrey-vl, 2018-03-05:
Well that's confusing since: 

> @seankinghan mbed CLI supports tags/branches to both add and import, e.g. using the the syntax mbed add/import <url>[#rev|branch|tag].

Anyway, from what I can tell using a branch name is indeed not supported, but wouldn't that make a nice feature?

theotherjimmy, 2018-03-05:
@geoffrey-vl That depends on what you expect `mbed add` with a branch name to do.

I could see branch name -> current hash in `.lib` file as a nice feature. OTOH, if you want to see branch name -> branch name in `.lib` file, I would veto that: I think it's a footgun as it's opposed to the reproducability that we generally go for. Besides, if you had branch -> hash in `.lib` you could later run `mbed update` to move to the head of the branch.

geoffrey-vl, 2018-03-06:
Yes I recon branch -> hash would be the better choice as well, that'd be nice and really convenient.

Just thinking out loud here, but maybe it could be the user's choice? Just like we do in Yocto, with Yocto you can use the autoinc option to auto fetch the latest revision. I think you'd only use this when targeting a branch that you control yourselves, otherwise you have no control at all on what get's pulled and so on.

screamerbg, 2018-05-26:
@sg- The suggestion is to enable #branch/tag specifier to `mbed add` and `mbed import`, e.g. `mbed import https://github.com/ARMmbed/sd-driver#master_update`.  The .lib files will contain the actual hash of the specified branch/tag name. What do you think?

closed 2019-10-31 by ciarmcom