ARMmbed/mbed-cli #580: allow sharing static libraries (esp. mbed-os) among multiple projects

bmcdonnell-ionx, 2017-12-04:
This is a feature(s) request, originally discussed here: [How can I factor mbed-os out of my (offline) project?](https://os.mbed.com/questions/79547/How-can-I-factor-mbed-os-out-of-my-offli/).

Currently, AFAIK, there is no way to **use a single copy of a static library from multiple projects**. I think it would be quite useful to be able to do this both entirely within `mbed-cli`, and also in exported projects (using offline IDEs). I think it would be especially useful to have `mbed-os` as a static library or libraries, which are also buildable as an exported static library project(s).

Suppose I have executable `ProjectBar`, which links to static library `LibFoo` (which could be mbed-os-lib). I'm requesting these features:

1. Ability to link to `LibFoo` within `ProjectBar`, while `LibFoo` resides outside the `ProjectBar` sourcetree. (i.e. Link to the already-built library and header files.)
   - `mbed-cli compile` links in `LibFoo`.
   - Can use relative and/or absolute paths.
2. Ability to export `LibFoo` to offline IDEs as a static library project.
3. When exporting `ProjectBar` to an offline IDE, maintain references to `LibFoo` in the project/linker files.
4. Ability to "step into" `LibFoo` source when debugging `ProjectBar`.

Of course, the last one is lowest priority relative to the others - kind of a "nice to have".

bmcdonnell-ionx, 2017-12-04:
Related: #547.

janjongboom, 2017-12-16:
@bmcdonnell-ionx So, one way would be to build the static library from a location, then link the static binary in through a symbolic link. Just a bit of work in initial setup, but after that it does what you want to do right?

bmcdonnell-ionx, 2017-12-16:
@janjongboom IT is heavily restricting local admin privileges on our PCs, so that would not be a practical solution for us.

https://github.com/git-for-windows/git/wiki/Symbolic-Links

Also, it would be preferable to have a group of repos that users can just clone and batch build. Clever, though.

bmcdonnell-ionx, 2018-01-24:
Let me draw a parallel here: consider MS Visual Studio solutions and projects.

# Solutions (the example)

Each project produces one output, be it an executable or a library. It is good practice to make projects modular. The solutions track dependencies between projects.

You could have a setup like this:

    $ ls -F MySuites/
    App1/
    App2/
    App3/
    libBar/
    libFoo/
    SuiteAll.sln
    SuiteMinimal.sln

`MySuites` is the top-level folder containing related solutions and projects. `App[123]` are projects that produce executables - each in their own folder. `lib{Foo,Bar}` are projects that produce libraries (say static libraries). The `SuiteAll.sln` solution builds all 3 `App`s; `SuiteMinimal.sln` builds just `App1`.

`App1` uses `libFoo`; `App2` uses `libBar`; `App3` uses both. etc.

Each *module* (executable or library project) only gets built once, and modules can reference other modules which *need not be in subfolders beneath the projects that reference them*.

# Mbed Land (the idea/proposal)

I think it would be great if `mbed-cli` could do this sort of thing - both from `mbed compile`, and also `mbed export`. For instance, to imitate the example above, maybe you'd have a top-level repo like this:

    $ ls -aF MySuites/
    .
    ..
    .git/
    app1.lib
    app2.lib
    app3.lib
    lib-bar.lib
    lib-foo.lib
    mbed-os.lib
    suite-all.json
    suite-minimal.json

Like it does now, `mbed import` would clone the repos referenced by all the `.lib` files. (Note, there's an overloading of the term "library" here - meaning either [1] a static library project (e.g. `lib-bar`), or [2] an Mbed `.lib` file that points to a repo (`lib-bar.lib` or `app1.lib`). Perhaps you'd come up with a different file extension.)

Anyway, passing one (or more?) of these `.json` files to `mbed compile` would build all the projects referenced in file(s), and link them according to their dependencies. (Perhaps `mbed build` would be a better name.) Likewise, passing one of these `.json` file(s) to `mbed export` would create the multiple projects, and whatever metadata tracks their dependencies (analog of the "solution") in the target IDE. That way you don't have multiple copies of these projects (`mbed-os`!) consuming disk space, and build time. AND you can still debug in the application and library code.

It sounds like a non-trivial effort, but I think it would be a leap and bound in terms of development scalability.

# Feedback

Mbed OS team: what do you think? Do you consider the current setup "good enough", where the whole `mbed-os` source tree gets replicated into every project? Do you have interest or plans to make this more flexible, or even do something like I've described here?

bridadan, 2018-01-24:
@bmcdonnell-ionx This almost works as you described as long as `libBar` and `libFoo` are git repositories.

All you have to do is the following:

```
cd App1
mbed add /absolute/path/to/libFoo
```
That will clone `libFoo` into the current `App1`. The downside is this:

1) requires you to use an absolute path, so it will only work on the current machine (as opposed to using remote repositories, where they are cloned over the network).
2) currently incorrectly saves the path with `https://` (ex. on my Windows machine the saved `.lib` file looks like `https://c//Users/bridan01/onedrive_arm/Documents/dev/lib_test/mylib/#e8ae428ee4d63433ac56c6218f557e0c458775db`)

If we were to solve at least #2, Would that be a workable solution to your issue?

bridadan, 2018-01-24:
Just as an update to my previous post, good news! `#2` is actually already fixed on master!

It's important to note, however, that if you make an update to a shared library in `App1`, this update won't be reflected in `App2` since that is a separate clone of the library repository. If you want it to be updated in `App2`, you should commit and push your changes to the shared library and then update the library inside `App2` by pulling down these changes.

The upside to this approach is that you can avoid breaking `App2` and `App3` and still experiment with changes inside the shared lib within `App1`. The downside is that you need to do a few more steps to keep all of these libraries in sync.

The build tools will follow symbolic links I believe, so if you _really_ want the source to be shared (and not have to do all the git commits) then you can do that. But just realize there is a bit more risk since there is less isolation.

bmcdonnell-ionx, 2018-01-24:
@bridadan, thanks for your input.

> This almost works as you described as long as libBar and libFoo are git repositories.

By "this", do you mean for single projects as described in [my OP](https://github.com/ARMmbed/mbed-cli/issues/580#issue-279103521)? Not a unified multi-project setup as described in [my later comment](https://github.com/ARMmbed/mbed-cli/issues/580#issuecomment-359996801) (a master top-level repo containing libs and "solutions"). Assuming so...

>     mbed add /absolute/path/to/libFoo

It doesn't seem to work, though. With `mbed-cli` 1.3.0, I tried copying an already-existing project to a new location (`test-proj/`). I deleted `mbed-os.lib` from it, moved the `mbed-os/` folder up to be a peer with `test-proj/`. When I executed your suggested `mbed add` command with `-vv`, I get

    [mbed] ERROR: Directory "C:\dev_temp\test01\test-proj\mbed-os" is not empty. Please ensure that the destination folder is empty.

> Would that be a workable solution to your issue?

If it works, it would certainly be a step forward. But is it intentional that it work, or just sort of "accidental"? I've not seen this documented anywhere as a feature, so I'd worry about it breaking. Also, relative paths (#`1`) would be preferred. How much work would it be to get that working?

However, I think I'd still much prefer my later-proposed more comprehensive idea ("solutions", etc.). I think this would help to keep things in sync, addressing issues like you posed in [your "update" comment](https://github.com/ARMmbed/mbed-cli/issues/580#issuecomment-360107241) (immediately preceding this comment of mine). What do you think of it? Can you answer my questions posed there ("feedback" section)?

theotherjimmy, 2018-01-24:
@bmcdonnell-ionx Why not just use a symlink? It seems to meet all of your requirements.

theotherjimmy, 2018-01-24:
Also, note that object files and static libraries are treated slightly differently WRT weak references by the linker, so we may not be able to package everything as a static library.

bmcdonnell-ionx, 2018-01-24:
> Why not just use a symlink?

[Answered above](https://github.com/ARMmbed/mbed-cli/issues/580#issuecomment-352163002):

> IT is heavily restricting local admin privileges on our PCs, so that would not be a practical solution for us.
>
> https://github.com/git-for-windows/git/wiki/Symbolic-Links
>
> Also, it would be preferable to have a group of repos that users can just clone and batch build. Clever, though.

Details: The Windows 10 Creators Update allows users to create symbolic links on Windows 10 without admin priviliges. Our IT department hasn't pushed it out yet. Alternatively, prior to the update, there is a way for domain admins to grant permission for certain people to use `mklink` without admin privs, but then using links further reduces portability, and complicates use for those without the capability. Also, some filesystems (e.g. FAT) don't support symbolic links, so you couldn't e.g. have such a working copy on a thumb drive.

Also, if it works as expected, using symlinks would present multiple copies of source files in an IDE, though they'd all be using the same location on disk. Not a huge deal, but sub-optimal, IMO, as the IDE presentation wouldn't reflect the project/library modularity. And there would be no way to build *just* a library in the IDE.

bmcdonnell-ionx, 2018-01-24:
@theotherjimmy,

> Also, note that object files and static libraries are treated slightly differently WRT weak references by the linker, so we may not be able to package everything as a static library.

Could you elaborate? How are they treated differently? What problems would it cause? etc.

bmcdonnell-ionx, 2018-01-24:
Addendum: using symlinks would also make the development environments more platform dependent, or requires some git hackery to make it play nice when checked out by multiple OSes.

https://stackoverflow.com/questions/5917249/git-symlinks-in-windows

theotherjimmy, 2018-01-24:
> Could you elaborate? How are they treated differently? What problems would it cause? etc.

Looking at [the ARM linker manual](http://www.keil.com/support/man/docs/armlink/armlink_pge1362065917715.htm), one sentence sticks out at me:

>The linker does not load an object from a library to resolve a weak reference. 

I'm not clear on how this would resolve weak references within libraries, or if it could even resolve weak references within libraries.

-----

Symlinks appear to be supported on Windows, [stating with Vista](https://msdn.microsoft.com/en-us/library/windows/desktop/aa365680(v=vs.85).aspx)

theotherjimmy, 2018-01-24:
> there is no way to use a single copy of a static library from multiple projects.

This is by design: the configuration system allows a project to change the contents of the library, which would prevent sharing or make sharing only possible with _exactly the same configuration_.

theotherjimmy, 2018-01-24:
In your multi-project workspace thingy, you could invoke the build tools directly and specify the paths to scan directly. For Example:
```bash
python2 mbed-os/tools/make.py -t GCC_ARM -m K64F --build BUILD/K64F/GCC_ARM --source App1 --source mbed-os --source libBar --source libFoo
```

bmcdonnell-ionx, 2018-01-24:
>> there is no way to use a single copy of a static library from multiple projects.
>
> This is by design: the configuration system allows a project to change the contents of the library, which would prevent sharing or make sharing only possible with *exactly the same configuration*.

I agree with the attempt to prevent developers from inadvertently using a version of a library that differs from what they reference in a project (in the `.lib` file), but "sharing...with exactly the same configuration" is exactly what I'm trying to do. So I respectfully disagree with the design decision - or perhaps just the implementation, if I'm missing something.

Isn't it a common use case to share an already-built library/RTOS among multiple firmware programs? If not, I'm surprised.

bmcdonnell-ionx, 2018-01-24:
> Looking at the [ARM linker manual](http://www.keil.com/support/man/docs/armlink/armlink_pge1362065917715.htm), one sentence sticks out at me:
>
>
> I'm not clear on how this would resolve weak [...]

Reading further on that page, I see that weak *references* are distinct from weak *definitions*. I'd be concerned if weak definitions didn't work. I'm not yet familiar with use cases for weak references. Does Mbed OS use them?

> In your multi-project workspace thingy, you could invoke the build tools directly and specify the paths to scan directly. For Example:
>
>     python2 mbed-os/tools/make.py -t GCC_ARM -m K64F --build BUILD/K64F/GCC_ARM --source App1 --source mbed-os --source libBar --source libFoo

This would require manually modifying the generated IDE project files, right? (Something I thought would be preferable to avoid, for maximum portability.) How would debugging work?

bridadan, 2018-01-25:
@bmcdonnell-ionx first of all, sorry for missing some of the details of your posts! I admit I skimmed them at first and I didn't get the full picture, my bad!

I think most of the pieces to solve this are already in place. Let's run through it together first so we're all on the same page. :
1. First, build mbed-os as a static library. To create the static library, I'm fairly certain you must use Mbed CLI and not an IDE (feel free to correct me @theotherjimmy):

```
git clone https://github.com/armmbed/mbed-os
cd mbed-os
mbed compile -m <your compile target> -t <your toolchain> --library
```

At this point you have a static library of mbed-os built with all of the necessary headers in `BUILD/libraries/mbed/<your compile target>/<your toolchain>`. Please note, if your IDE uses GCC ARM as its compiler, you should use that when building the static library Mbed CLI.

2. Now what we need to do is create a new project:

```
cd ../
mbed new app1 --create-only
```

This will create an mbed project without cloning `mbed-os`, but still pull in the tools to compile the app.

3. Add some application code in `main.cpp` (I pulled the source for [mbed-os-example-blinky](https://github.com/ARMmbed/mbed-os-example-blinky/blob/master/main.cpp))

4. Export to your ide

```
cd app1
mbed export -m <your compile target> -i <your ide> --source . --source ../mbed-os/BUILD/libraries/mbed/<your compile target>/<your toolchain>
```

I tried this with IAR workbench and it worked fine. Now, I admit I tried this with the `make_gcc_arm` exporter and it threw an error when linking, however this may have been because of file path length on Windows.

bridadan, 2018-01-25:
I should also note that while this workflow works as of right now, it is by no means guaranteed to work in the future. Mbed OS is typically designed to work seamlessly in applications *only* when being compiled from source. If you do these steps, the "pieces of the puzzle" generally work together but it is not really a supported flow, so please use this at your own risk! I don't believe we plan on adding full support for this kind flow in the near feature due to the concerns @theotherjimmy mentioned above, the corner cases just get too big to support efficiently (also feel free to correct me on that @theotherjimmy).

0xc0170, 2018-01-25:
> Reading further on that page, I see that weak references are distinct from weak definitions. I'd be concerned if weak definitions didn't work. I'm not yet familiar with use cases for weak references. Does Mbed OS use them?

yes it uses, but we are building from source. The problem as mentioned earlier is with archives, known problem. Therefore for mbed 2 libraries, some files are object files that have weak symbols in for instance mbed retarget code file (see the code that handles it https://github.com/ARMmbed/mbed-os/blob/master/tools/build_api.py#L1043)

bmcdonnell-ionx, 2018-01-25:
@0xc0170, the code you linked to says it's copying the object files, so isn't that working around the fact that the weak references aren't in the built library? Such that you could use it as a built static library + a few object files.

I'm probably misunderstanding something, but if you're just building everything into a single executable output/object, I don't understand the need to copy those few object files.

bmcdonnell-ionx, 2018-01-25:
@bridadan,

Thanks for taking another look.

> I think most of the pieces to solve this are already in place...

> I should also note that while this workflow works as of right now, it is by no means guaranteed to work in the future... If you do these steps, the "pieces of the puzzle" generally work together but it is not really a supported flow, so please use this at your own risk!

Not a path I'm keen to follow, then. So here I am asking for new features. 😄

> Mbed OS is typically designed to work seamlessly in applications only when being compiled from source.

Back to the fundamental questions, then: *Does the Mbed OS team think that this currently-supported workflow is sufficient?*

> I don't believe we plan on adding full support for this kind flow in the near feature due to the concerns @theotherjimmy mentioned above, the corner cases just get too big to support efficiently (also feel free to correct me on that @theotherjimmy).

Is my use case unusual? e.g. If a design/development team is experimenting by writing multiple applications for a device. Or maybe they sell products that run on the same platform, but are differentiated in part by their application firmwares.

Don't you even bump up against this yourselves, during development, debugging, troubleshooting, etc.? How many projects do you have working copies of on your machine right now? With a copy of the whole `mbed-os` source tree for every one? Am I doing something fundamentally wrong or different here? Am I missing some obvious way to achieve my desired decoupling and reuse? (Besides the symlink hack.)

theotherjimmy, 2018-01-25:
>Don't you even bump up against this yourselves, during development, debugging, troubleshooting, etc.? 

All the time. I use symlinks, and don't consider them a hack.

bmcdonnell-ionx, 2018-01-25:
@theotherjimmy, do you commit the symlinks to your repos as symlinks? Or do you always have to `git clone`, `cd newprojdir`, `ln -s ../mbed-os .`, `mbed deploy` / `mbed export` etc., every time you grab a new application repo?

If there are actually symlinks in your repos, do they work across development platforms? (Windows/Linux/Mac)

theotherjimmy, 2018-01-25:
@bmcdonnell-ionx I do not commit the symlinks.

I also don't use `mbed deploy` or `mbed export`. `mbed deploy` does not suite a use case where you develop mbed-os, and `mbed export` is not useful when developing the build tools.

bmcdonnell-ionx, 2018-01-26:
@0xc0170,

> yes [Mbed OS] uses [weak references], but we are building from source. The problem as mentioned earlier is with archives, known problem. Therefore for mbed 2 libraries, some files are object files that have weak symbols in for instance mbed retarget code file (see the code that handles it https://github.com/ARMmbed/mbed-os/blob/master/tools/build_api.py#L1043)

I thought [compiling static libraries](https://github.com/ARMmbed/mbed-cli#compiling-static-libraries) is a supported feature...?

theotherjimmy, 2018-01-26:
@bmcdonnell-ionx Is that bit of instructions not working for you?

bmcdonnell-ionx, 2018-01-29:
@theotherjimmy, what I mean is you seem to be saying here that compiling Mbed OS as a static library is not a supportable feature, based on the lack of support for weak references in Keil's compiler. But I'm pointing out that it *is* apparently a supported feature - though only for the Mbed CLI builder - as indicated by the reference I linked.

However, I'm here asking for this support for the static library feature with the Mbed CLI compiler to extend to `mbed export`, and to the use of the static library. How does one use the built Mbed OS static library once built? In an exported project. Without having to redo manual steps every time the need to re-export arises?

theotherjimmy, 2018-01-29:
@bmcdonnell-ionx I think that I'm saying that we don't test "Mbed OS as a static library", and the tools are tested for building static libraries. So `--library` works, but may not work with Mbed OS. OTOH, `mbed compile --library` supports an additional `--no-archive` flag, that disables creation of the `.a` file. That **will** work with Mbed OS, as it will use the `.o` files from the build, without rebuilding them.

----------------

`mbed export` is designed to export **projects**, and is not capable of `--library` style exports. You might be able to do something with placing the `--library --no-archive` build in tree, then exporting. I do not test the command line version of that (the website does something similar). I would be quite willing to modify `mbed export --source ... --source ...` behavior to work well with `mbed compile --library --no-archive` as that would make specifying multiple sources a useful command line feature. Let me know on that one.

amq, 2018-08-19:
`mbed cache on` solved this for me.

bmcdonnell-ionx, 2018-08-20:
@amq, I don't see any connection between `mbed cache` and my feature request here.

j3270, 2018-09-17:
#646 

I think the issue above hurts users like me and op.  I haven't been as active with mbed lately, so forgive me if either of these issues has been addressed.  I know #646 was closed, but don't think it should have been.

If we --create-only, we still need the tools to build the app we are working on and reference our archives; mbed-os, lib1, lib2,etc...  Hope that made sense.

Thanks,

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2019-10-31 by ciarmcom