ARMmbed/mbed-cli #581: [ERROR] ./main.cpp:1:18: fatal error: mbed.h: No such file or directory

wangnuannuan, 2017-12-08:
I try to modify the mbed.py ,so i can use the mbed-cli offline.what i do is copy the mbed-os to a folder:sys.executable(/usr/bin/)/mbedcli/mbed-os. 

first for the "mbed compile XXXX" : I change the **get_os_dir()**
    def get_os_dir(self):
        '''if os.path.isdir(os.path.join(self.path, 'mbed-os')):
            return os.path.join(self.path, 'mbed-os')'''
        if os.path.isdir(os.path.join(mbed_os_local_abspath,'mbed-os')):
            return os.path.join(mbed_os_local_abspath,'mbed-os')
then when i run "mbed compile -m K64F -t GCC_ARM"
Building project mbed-os-example-blinky (K64F, GCC_ARM)
Scan: .
Scan: mbed
Scan: env
Compile [ 50.0%]: main.cpp
[Fatal Error] main.cpp@1,18: mbed.h: No such file or directory
[ERROR] ./main.cpp:1:18: fatal error: mbed.h: No such file or directory
 #include "mbed.h"
                  ^
compilation terminated.

[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /usr/bin/mbedcli/mbed-os/tools/make.py -t GCC_ARM -m K64F --source . --build ./BUILD/K64F/GCC_ARM" in "/home/nuan/Documents/mbed-os-example-blinky"
---

theotherjimmy, 2017-12-08:
Hi @wangnuannuan Thanks for reporting this! I updated your comment to use ` ``` ` for code formatting, which is required by github.

theotherjimmy, 2017-12-08:
@wangnuannuan Mbed CLI tries to make it clear when it will access the network, and should work when it does not have network access.

Commands that will use the network:
* `mbed new`       Create new mbed program or library
* `mbed import`    Import program from URL
* `mbed add`        Add library from URL
* `mbed deploy`     Find and add missing libraries
* `mbed publish`    Publish program or library
* `mbed update`     Update to branch, tag, revision or latest
* `mbed sync`       Synchronize library references

Other commands do not use any network activity, and will work offline.

What exactly are you trying to do?

theotherjimmy, 2017-12-08:
Sorry about the spurious close. I miss-clicked.

wangnuannuan, 2017-12-08:
every time when i run '''mbed new'''or '''mbed import''. It takes long time to download the 'mbed-os',so  i change the code adding the mbed-os folder in the folder mbed-cli,and then  i install mbed-cli,the mbed-os will be copy to a fixed path. when run '''mbed compile'''  ,the above error occurred

theotherjimmy, 2017-12-08:
@wangnuannuan You could turn on caching. You could also just git/hg clone the repo followed by creating a symlink to a copy of Mbed OS matching the `.lib` file. That error happens because Mbed OS is not actually in your project.

wangnuannuan, 2017-12-08:
but i think that if my computer is ofline ,then i can't create a new project, i have to copy mbed-os to my project ,I think it's a bit of a hassle

theotherjimmy, 2017-12-08:
@wangnuannuan That's why I recommend using a symlink. It should be trivially fast, and take almost no extra disk space.

wangnuannuan, 2017-12-09:
thanks for your recommendation，i have solved the error

closed 2018-01-12 by cmonr