ARMmbed/mbed-cli #583: mbed-cli goes to application main root when calling mbed deploy from sub-folder

teetak01, 2017-12-11:
**Bug:**
Critical

**Expected behaviour:**

mbed deploy will deploy only the libraries in current folders and its subfolders.

**Actual behavior:**

mbed deploy will go to application root and start redeploying the whole application tree

**Steps to reproduce:**
```
mbed import mbed-os-example-client
mkdir tmp
cp mbed-client.lib tmp
cd tmp
mbed deploy
```
```
~/devel/mbed-os-example-client (master) $ cp mbed-client.lib tmp
~/devel/mbed-os-example-client (master) $ cd tmp
~/devel/mbed-os-example-client/tmp (master) $ mbed deploy --verbose
[mbed] Working path "/home/xxx/devel/mbed-os-example-client/tmp" (program)
[mbed] Updating library "-os" to rev #5f6572179d66
[mbed] Fetching revisions from remote repository to "mbed-os"
```

This behaviour is highly problematic as I may very well have different versions of libraries in my development tree and I definately do not want to touch those.

Also the tracing is quite broken in this case as mbed-cli is not able to print the module-names correctly.

bridadan, 2017-12-11:
Hi @teetak01, thanks for filing such a detailed report! I was able to reproduce the situation. Let's go through it.

> mbed deploy will deploy only the libraries in current folders and its subfolders.

This is almost true. `mbed deploy` will only deploy libraries that are "below" the current library/application. This is done based on repositories. It will look for the nearest parent repository. Since in this case that is the application, it is in fact doing the correct thing, however I can understand how that's confusing.

Your particular use case is interesting, however I think you'd be better served by the `mbed add` command. Just copy the contents of your `.lib` into the command: `mbed add <contents of .lib here>`.

> This behaviour is highly problematic as I may very well have different versions of libraries in my development tree and I definately do not want to touch those.

I don't quite understand the problem, you shouldn't have multiple copies of libraries in the same repository as that will cause compilation/linking errors.

> Also the tracing is quite broken in this case as mbed-cli is not able to print the module-names correctly.

You are correct! I think I've tracked down what's causing the strange tracing, not sure if my fix is the correct way to go about it. I'll be posting a proposal PR shortly.

teetak01, 2017-12-12:
Hi @bridadan I think I understand this issue now a bit better. This is relatively corner-case what I am having.

If I need to deploy a single .lib file, I really do not have any straightforward way. Having something like 

`mbed deploy mylib.lib` actually deploying only the given lib-file would be perfect. Instead we have to read the contents of .lib file and perform `mbed add` as you mentioned.

Well, I think that is probably what I have to do. I just just quite unintuitive that mbed deploy looks for the current repository root and runs the command there, not where it was called.

teetak01, 2017-12-12:
Anyways, #584 is ok enough for this, unless you are willing to reconsider adding option to deploy existing lib-files directly.

bridadan, 2017-12-12:
I'm still a bit confused of your use case though. If you already have the .lib file present, you must know the URL you would otherwise be passing to `mbed add`. Do you have a repo I can look at where you're doing this? I'm a bit worried that we already have a mechanism to solve your problem and we may be over-complicating the issue 😄

teetak01, 2017-12-15:
This came up in relative abnormal use-case (internal release generation), where I need to pick specific components from several libraries. Although I have also encountered this occationally in other use-cases too, often as part of CI-automation, which also often has to do some tricks to get the desired end result. Well, I have functional workaround for this, so this can be closed as part of #584.

teetak01, 2017-12-21:
Ok, thanks @bridadan and @screamerbg

closed 2017-12-21 by teetak01