ARMmbed/mbed-cli #585: Windows installer does not provide mbed cli on the path

robotdad, 2017-12-13:
I just installed mbed_installer_v043.exe with the defaults. The command mbed is not available on the command prompt or in powershell. It is not clear what should be set to fix this as the only thing in the install directory are python scripts. I see nothing in the installation that would provide the mbed command used in all of the documentation.

robotdad, 2017-12-13:
OK, it looks like it installed it under my Python 2.7 install that was already present. I have to run mbed from within the scripts directory of the python 2.7 install location, it is not available on the system path.

robotdad, 2017-12-14:
And now trying to use that it fails to automatically install missing modules, so I have to install each by hand as I hit them. It also does not have the path to the arm-gcc compiler set etc.

screamerbg, 2017-12-20:
cc @arekzaluski

screamerbg, 2018-04-28:
Bump @arekzaluski