ARMmbed/mbed-cli #588: export multiple profiles as multiple build configurations

bmcdonnell-ionx, 2017-12-20:
### Feature request

Add ability to export multiple build profiles as multiple build configurations in a single IDE project.

Does this belong here or under mbed-os?

### Background

Inspired by [C++11 in exported projects](https://os.mbed.com/questions/79757/C11-in-exported-projects/?c=29224).

In source control for my projects, I'm currently keeping just the files needed so that a developer can check out, and either build or export with mbed-cli. This is in keeping with the convention of publicly-available sample programs.

`mbed export` apparently creates multiple build configurations - one each for Debug and Release - at least it does for MCUXpresso. (`mbed export -m LPC4088 -i mcuxpresso`.) But AFAIK, there is no way to both customize build flags (e.g. `-std=gnu++11`), and have more than one build configurations in the exported project.

screamerbg, 2017-12-20:
cc @theotherjimmy

screamerbg, 2018-04-28:
Bump @theotherjimmy

theotherjimmy, 2018-04-30:
@bmcdonnell-ionx I have no plans to implement multiple profile support at this time.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

bmcdonnell-ionx, 2018-10-02:
> *Issue header has missing or incorrectly filled in fields! Please fix.*

@adbridge, what does that mean?

adbridge, 2018-10-02:
@bmcdonnell-ionx We have recently added an issue pro-forma which requires a checkbox selection stating the type of issue, to be filled in. There is also a new automated script which checks this which has just been enabled. It looks like mbed-cli does not have that proforma currently and this issue was raised prior to that coming in. I will fix it :)

closed 2019-10-31 by ciarmcom