ARMmbed/mbed-cli #589: Build for *_boot target fails with AddressOverlapError: Starting addresses are different

vkolotov, 2017-12-21:
Hi All,

I'm trying to build my projects locally with mbed-cli tool for a *_boot target. I can do this successfully via online compiler, but mbed-cli fails with "AddressOverlapError: Starting addresses are different".

I'm also able to build for *_ota and just without any postfixes, e.g.: 
```
mbed compile -t GCC_ARM -m nrf51822_ota -c
```
and
```
mbed compile -t GCC_ARM -m nrf51822 -c
```

Here are some details:
```
mbed compile -t GCC_ARM -m nrf51822_boot -c -v

....

[DEBUG] Errors: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-sbrkr.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
[DEBUG] Errors: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/bin/ld: warning: /usr/lib/gcc/arm-none-eabi/4.9.3/../../../arm-none-eabi/lib/armv6-m/libc_nano.a(lib_a-nano-msizer.o) uses 2-byte wchar_t yet the output is to use 4-byte wchar_t; use of wchar_t values across objects may fail
Elf2Bin: SmartLock
[DEBUG] FromELF: /usr/bin/arm-none-eabi-objcopy -O ihex ../BUILD/nrf51822_boot/GCC_ARM/SmartLock.elf ../BUILD/nrf51822_boot/GCC_ARM/SmartLock.hex
[DEBUG] Return: 0
[DEBUG] SoftDevice file found s130_nrf51_1.0.0_softdevice.hex.
[DEBUG] Bootloader file found s130_nrf51_1.0.0_bootloader.hex.
[DEBUG] Merge SoftDevice file s130_nrf51_1.0.0_softdevice.hex
[DEBUG] Merge BootLoader file ../BLE_API/nRF51822/bootloader/s130_nrf51_1.0.0_bootloader.hex
Traceback (most recent call last):
  File "/home/meccup/projects/sputnikdev/SmartLock/.temp/tools/make.py", line 296, in <module>
    stats_depth=options.stats_depth)
  File "/home/meccup/projects/sputnikdev/SmartLock/.temp/tools/build_api.py", line 592, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "/home/meccup/projects/sputnikdev/SmartLock/.temp/tools/toolchains/__init__.py", line 1146, in link_program
    self.binary(r, elf, bin)
  File "/home/meccup/projects/sputnikdev/SmartLock/.temp/tools/hooks.py", line 54, in wrapper
    post_res = tooldesc["post"](t_self, *args, **kwargs)
  File "/home/meccup/projects/sputnikdev/SmartLock/.temp/tools/targets/__init__.py", line 519, in binary_hook
    binh.merge(blh)
  File "/usr/local/lib/python2.7/dist-packages/intelhex/__init__.py", line 853, in merge
    'Starting addresses are different')
AddressOverlapError: Starting addresses are different
[mbed] ERROR: "/usr/bin/python" returned error code 1.
[mbed] ERROR: Command "/usr/bin/python -u /home/meccup/projects/sputnikdev/SmartLock/.temp/tools/make.py -t GCC_ARM -m nrf51822_boot --source .. --build ../BUILD/nrf51822_boot/GCC_ARM -c -v" in "/home/meccup/projects/sputnikdev/SmartLock/src"
---

```

Am I doing something wrong?
Cheers

screamerbg, 2017-12-21:
cc @theotherjimmy @pan-

screamerbg, 2018-04-28:
@vkolotov Have you been able to find solution to your question?

melvinvdb, 2018-05-09:
I'm having the same issue.
Reproduces using the ble_led example with command
`mbed compile --target NRF51_DK_BOOT --toolchain GCC_ARM -v`

results in:
`[DEBUG] SoftDevice file found s130_nrf51_1.0.0_softdevice.hex.
[DEBUG] Bootloader file found s130_nrf51_1.0.0_bootloader.hex.
[DEBUG] Merge SoftDevice file s130_nrf51_1.0.0_softdevice.hex
[DEBUG] Merge BootLoader file ./mbed-os/features/FEATURE_BLE/targets/TARGET_NORDIC/TARGET_MCU_NRF51822/bootloader/s130_nrf51_1.0.0_bootloader.hex
Traceback (most recent call last):
  File "/Users/melvinvandenberg/Downloads/mbed-os-example-ble/BLE_LED/mbed-os/tools/make.py", line 298, in <module>
    stats_depth=options.stats_depth)
  File "/Users/melvinvandenberg/Downloads/mbed-os-example-ble/BLE_LED/mbed-os/tools/build_api.py", line 589, in build_project
    res, _ = toolchain.link_program(resources, build_path, name)
  File "/Users/melvinvandenberg/Downloads/mbed-os-example-ble/BLE_LED/mbed-os/tools/toolchains/__init__.py", line 1154, in link_program
    self.binary(r, elf, bin)
  File "/Users/melvinvandenberg/Downloads/mbed-os-example-ble/BLE_LED/mbed-os/tools/hooks.py", line 54, in wrapper
    post_res = tooldesc["post"](t_self, *args, **kwargs)
  File "/Users/melvinvandenberg/Downloads/mbed-os-example-ble/BLE_LED/mbed-os/tools/targets/__init__.py", line 521, in binary_hook
    binh.merge(blh)
  File "/Library/Python/2.7/site-packages/intelhex/__init__.py", line 872, in merge
    'Starting addresses are different')
AddressOverlapError: Starting addresses are different`

This happens after I fixed bug [https://github.com/ARMmbed/mbed-os/issues/6835](https://github.com/ARMmbed/mbed-os/issues/6835)

Using mbed-cli 1.5.1 on Mac OS

theotherjimmy, 2018-05-09:
@screamerbg I can reproduce this issue. I'll look into it while I'm thinking about it.

melvinvdb, 2018-06-05:
Changing `binh.merge(blh)` to `binh.merge(blh, overlap='replace')` in mbed-os/tools/targets/__init__.py seems to solve this issue.
It is kind of weird because the hex files do not overlap (checked in hex editor).

theotherjimmy, 2018-06-05:
@melvinvdb Yeah, it might be that both files have a start address, which will conflict and generate overlap errors.

screamerbg, 2018-06-18:
@theotherjimmy What's the conclusion on this? Can we close the issue here?

theotherjimmy, 2018-06-19:
https://github.com/ARMmbed/mbed-os/pull/7263

cmonr, 2018-06-20:
@vkolotov @melvinvdb @screamerbg The fix has now been merged. Would y'all mind retesting?

theotherjimmy, 2018-09-24:
I can't reproduce the errors anymore

closed 2018-09-24 by theotherjimmy