ARMmbed/mbed-cli #594: mbed releases not working on Windows

JanneKiiskila, 2017-12-27:
Seems the command `mbed releases` is not working with the Windows version of this. I have installed the whole Mbed OS toolchain using the Windows installer we have.

```
C:\Users\jankii01\mbed\mbed-os-example-client>mbed ls
mbed-os-example-client (5767847be0d2)
|- easy-connect (522e43fa1bba)
|  |- atmel-rf-driver (2cd9abba3e37)
|  |- esp8266-driver (a7245780485c)
|  |- mcr20a-rf-driver (9aac10474702)
|  |- stm-spirit1-rf-driver (0ff4ca7537f0)
|  `- wifi-x-nucleo-idw01m1 (537b01c4a053)
|- mbed-client (ea04c5de7822)
|  |- mbed-client-c (ecfa619e42b2)
|  |- mbed-client-classic (4e66929607c3)
|  `- mbed-client-mbed-tls (7e1b6d815038)
|- mbed-os (da9464d10305)
`- pal (60ce64d5ec35)

C:\Users\jankii01\mbed\mbed-os-example-client>mbed releases
usage: mbed [-h] [--version]             ...
mbed: error: argument            : invalid choice: 'releases' (choose from 'new', 'import', 'add', 'remove', 'deploy', 'publish', 'update', 'sync', 'ls', 'status', 'compile', 'test', 'export', 'detect', 'config', 'target', 'toolchain', 'help')

C:\Users\jankii01\mbed\mbed-os-example-client>mbed --version
1.2.2
```
As per documentation the `mbed releases` should list the version numbers (if available) from each repo.

JanneKiiskila, 2017-12-27:
Okay, some guess work - the mbed releases command was added to version 1.3 and I am now guessing that's not coming through to Windows (yet). The installer you link to - gives you only version 1.1.1.

https://github.com/ARMmbed/mbed-cli/tree/1.3.0#installing-mbed-cli

If you re-do the `pip install mbed-cli` - it only also gives you the 1.2.2.

```
C:\Users\jankii01\mbed\mbed-cloud-client-example-internal>pip install mbed-cli
Requirement already satisfied: mbed-cli in c:\python27\lib\site-packages\mbed_cli-1.2.2-py2.7.egg
```

JanneKiiskila, 2017-12-27:
However, doing `pip download` found 1.3.0. As did `pip search`.
I did an `pip uninstall` - then it failed removing it.

```
C:\Users\jankii01\mbed\mbed-cloud-client-example-internal>pip uninstall mbed-cli
Uninstalling mbed-cli-1.2.2:
  c:\python27\lib\site-packages\mbed_cli-1.2.2-py2.7.egg
  c:\python27\scripts\mbed-cli-script.py
  c:\python27\scripts\mbed-cli.exe
  c:\python27\scripts\mbed-cli.exe.manifest
  c:\python27\scripts\mbed-script.py
  c:\python27\scripts\mbed.exe
  c:\python27\scripts\mbed.exe.manifest
Proceed (y/n)? y
  Successfully uninstalled mbed-cli-1.2.2
Traceback (most recent call last):
  File "c:\python27\lib\runpy.py", line 174, in _run_module_as_main
    "__main__", fname, loader, pkg_name)
  File "c:\python27\lib\runpy.py", line 72, in _run_code
    exec code in run_globals
  File "C:\Python27\Scripts\pip.exe\__main__.py", line 9, in <module>
  File "c:\python27\lib\site-packages\pip\__init__.py", line 233, in main
    return command.main(cmd_args)
  File "c:\python27\lib\site-packages\pip\basecommand.py", line 252, in main
    pip_version_check(session)
  File "c:\python27\lib\site-packages\pip\utils\outdated.py", line 102, in pip_version_check
    installed_version = get_installed_version("pip")
  File "c:\python27\lib\site-packages\pip\utils\__init__.py", line 838, in get_installed_version
    working_set = pkg_resources.WorkingSet()
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 644, in __init__
    self.add_entry(entry)
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 700, in add_entry
    for dist in find_distributions(entry, True):
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1949, in find_eggs_in_zip
    if metadata.has_metadata('PKG-INFO'):
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1463, in has_metadata
    return self.egg_info and self._has(self._fn(self.egg_info, name))
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1823, in _has
    return zip_path in self.zipinfo or zip_path in self._index()
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1703, in zipinfo
    return self._zip_manifests.load(self.loader.archive)
  File "c:\python27\lib\site-packages\pip\_vendor\pkg_resources\__init__.py", line 1643, in load
    mtime = os.stat(path).st_mtime
WindowsError: [Error 2] The system cannot find the file specified: 'c:\\python27\\lib\\site-packages\\mbed_cli-1.2.2-py2.7.egg'
```

But, after that doing `pip install mbed-cli` brought me up to version 1.3.0. Something not quite working with `pip` here, I think.

closed 2017-12-27 by JanneKiiskila

screamerbg, 2017-12-27:
The correct command to upgrade is `pip install -U mbed-cli`

JanneKiiskila, 2017-12-27:
We should probably give that info in the `README.md` then.

JanneKiiskila, 2017-12-27:
https://github.com/ARMmbed/mbed-cli/pull/595