ARMmbed/mbed-cli #596: mbed releases -r - listing too long

JanneKiiskila, 2017-12-29:
Why is the tool listing **all** versions for the components? The list is too long.. I would understand if you used something like `--all` to get ALL the versions, but for now - it should just list ONE version (the one that's used) OR the hash (if no version match).

JanneKiiskila, 2017-12-29:
@screamerbg

screamerbg, 2017-12-29:
@sg- has a similar feedback. Will look into it

screamerbg, 2018-04-28:
@JanneKiiskila We've recently added wider release tags awareness to Mbed CLI.

>  it should just list ONE version (the one that's used) OR the hash (if no version match).

Mbed CLI 1.5.1 now shows the release tags per library with the `mbed ls` command, e.g.
```
$ mbed ls
mbed-os-example-client (#98251b8e4bab)
|- easy-connect (#6b2cce9f7cef, tag: v1.2.2)
|  |- atmel-rf-driver (#ca9782e68f5f)
|  |- esp8266-driver (#c2b7e15c08ca)
|  |- mcr20a-rf-driver (#93661a696735)
|  |- stm-spirit1-rf-driver (#ce9e2f81f95f, tag: v1.0.1)
|  |- wifi-ism43362 (#d47a8c2fba0e)
|  |- wifi-x-nucleo-idw01m1 (#257d0878561b)
|  `- wizfi310-driver (#e0f7b9355e7e)
|- mbed-client (#ea04c5de7822)
|  |- mbed-client-c (#ecfa619e42b2, tag: v6.0.0)
|  |- mbed-client-classic (#4e66929607c3)
|  `- mbed-client-mbed-tls (#7e1b6d815038, tags: mbedCloudClient-R1.1, mbedCloudClient-R1.1-RC1)
|- mbed-os (#c05d72c3c005, tags: latest, mbed-os-5.8.3)
`- pal (#60ce64d5ec35)
```

In the case of Mbed OS and mbed-client-mbed-tls, it will list multiple tags if detect.

Does this solve your suggestion?