ARMmbed/mbed-cli #597: mbed detect doesn't detect target unless using verbose mode

MarceloSalazar, 2018-01-02:
Summary:
`mbed detect` does not show output when invoked without arguments, but does work when using `-vv`
Host OS: Windows 10 on VM (under OSX).
Version: 1.3.0

Log:

```
C:\mbed-os-example-blinky>mbedls
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| platform_name | platform_name_unique | mount_point | serial_port | target_id                                        | daplink_version |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+
| K64F          | K64F[0]              | E:          | COM3        | 0240000030514e45005b20067d7e001c1f91000097969900 | 0244            |
+---------------+----------------------+-------------+-------------+--------------------------------------------------+-----------------+

C:\mbed-os-example-blinky>mbed detect

C:\mbed-os-example-blinky>mbed detect -v
[mbed] Working path "C:\mbed-os-example-blinky" (program)

C:\mbed-os-example-blinky>mbed detect -vv
[mbed] Working path "C:\mbed-os-example-blinky" (program)
[mbed] Query "c:\python27\python.exe -u C:\mbed-os-example-blinky\mbed-os\tools\detect_targets.py" in C:\mbed-os-example-blinky
[mbed] Detected K64F, port COM3, mounted E:, interface version 0244:
[mbed] Supported toolchains for K64F
+--------+-----------+-----------+-----------+-----------+-----------+-----------+
| Target | mbed OS 2 | mbed OS 5 |    ARM    |  GCC_ARM  |    IAR    |   ARMC6   |
+--------+-----------+-----------+-----------+-----------+-----------+-----------+
| K64F   | Supported | Supported | Supported | Supported | Supported | Supported |
+--------+-----------+-----------+-----------+-----------+-----------+-----------+
Supported targets: 1
Supported toolchains: 4

C:\mbed-os-example-blinky>mbed --version
1.3.0
```

MarceloSalazar, 2018-01-05:
I'm also seeing this problem natively on OSX.
IMO, this is a "major" regression issue. **Note** it doesn't happen on v1.2.2.

FYI @bridadan @screamerbg @theotherjimmy

screamerbg, 2018-02-06:
This is now merged and released in Mbed CLI 1.4.0

closed 2018-02-06 by screamerbg