ARMmbed/mbed-cli #598: Cannot install on fresh Mac with High Sierra

janjongboom, 2018-01-04:
Got a new Mac today. Installation of Mbed CLI fails when trying to fix the mbed-os requirements:

```
janjon01:~/repos/mbed-os-example-blinky/mbed-os (mbed-os-5.7) $ sudo pip install -r requirements.txt
The directory '/Users/janjon01/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/janjon01/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: colorama>=0.3.3 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 1))
Requirement already satisfied: PySerial>=2.7 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 2))
Requirement already satisfied: PrettyTable>=0.7.2 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 3))
Requirement already satisfied: Jinja2>=2.7.3 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 4))
Requirement already satisfied: IntelHex>=1.3 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 5))
Requirement already satisfied: junit-xml in /Library/Python/2.7/site-packages (from -r requirements.txt (line 6))
Requirement already satisfied: pyYAML in /Library/Python/2.7/site-packages (from -r requirements.txt (line 7))
Requirement already satisfied: requests in /Library/Python/2.7/site-packages (from -r requirements.txt (line 8))
Requirement already satisfied: mbed-ls>=0.2.13 in /Library/Python/2.7/site-packages (from -r requirements.txt (line 9))
Collecting mbed-host-tests>=1.1.2 (from -r requirements.txt (line 10))
  Downloading mbed-host-tests-1.3.0.tar.gz (94kB)
    100% |████████████████████████████████| 102kB 1.7MB/s
Collecting mbed-greentea>=0.2.24 (from -r requirements.txt (line 11))
  Downloading mbed-greentea-1.3.2.tar.gz (94kB)
    100% |████████████████████████████████| 102kB 5.9MB/s
Collecting beautifulsoup4>=4 (from -r requirements.txt (line 12))
  Downloading beautifulsoup4-4.6.0-py2-none-any.whl (86kB)
    100% |████████████████████████████████| 92kB 6.1MB/s
Collecting fuzzywuzzy>=0.11 (from -r requirements.txt (line 13))
  Downloading fuzzywuzzy-0.16.0-py2.py3-none-any.whl
Requirement already satisfied: MarkupSafe>=0.23 in /Library/Python/2.7/site-packages (from Jinja2>=2.7.3->-r requirements.txt (line 4))
Requirement already satisfied: six in /System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python (from junit-xml->-r requirements.txt (line 6))
Requirement already satisfied: certifi>=2017.4.17 in /Library/Python/2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /Library/Python/2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already satisfied: idna<2.7,>=2.5 in /Library/Python/2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already satisfied: urllib3<1.23,>=1.21.1 in /Library/Python/2.7/site-packages (from requests->-r requirements.txt (line 8))
Requirement already satisfied: fasteners in /Library/Python/2.7/site-packages (from mbed-ls>=0.2.13->-r requirements.txt (line 9))
Requirement already satisfied: appdirs>=1.4 in /Library/Python/2.7/site-packages (from mbed-ls>=0.2.13->-r requirements.txt (line 9))
Collecting pyOCD>=0.8.1a1 (from mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
  Downloading pyOCD-0.8.1a1.tar.gz (441kB)
    100% |████████████████████████████████| 450kB 3.1MB/s
Collecting lockfile (from mbed-greentea>=0.2.24->-r requirements.txt (line 11))
  Downloading lockfile-0.12.2-py2.py3-none-any.whl
Collecting mock (from mbed-greentea>=0.2.24->-r requirements.txt (line 11))
  Downloading mock-2.0.0-py2.py3-none-any.whl (56kB)
    100% |████████████████████████████████| 61kB 6.5MB/s
Requirement already satisfied: monotonic>=0.1 in /Library/Python/2.7/site-packages (from fasteners->mbed-ls>=0.2.13->-r requirements.txt (line 9))
Requirement already satisfied: enum34 in /Library/Python/2.7/site-packages (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already satisfied: future in /Library/Python/2.7/site-packages (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Requirement already satisfied: websocket-client in /Library/Python/2.7/site-packages (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
Collecting hidapi (from pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
  Downloading hidapi-0.7.99.post21.tar.gz (51kB)
    100% |████████████████████████████████| 61kB 8.2MB/s
Collecting pbr>=0.11 (from mock->mbed-greentea>=0.2.24->-r requirements.txt (line 11))
  Downloading pbr-3.1.1-py2.py3-none-any.whl (99kB)
    100% |████████████████████████████████| 102kB 4.7MB/s
Collecting funcsigs>=1; python_version < "3.3" (from mock->mbed-greentea>=0.2.24->-r requirements.txt (line 11))
  Downloading funcsigs-1.0.2-py2.py3-none-any.whl
Collecting setuptools>=19.0 (from hidapi->pyOCD>=0.8.1a1->mbed-host-tests>=1.1.2->-r requirements.txt (line 10))
  Downloading setuptools-38.2.5-py2.py3-none-any.whl (489kB)
    100% |████████████████████████████████| 491kB 2.3MB/s
Installing collected packages: setuptools, hidapi, pyOCD, mbed-host-tests, lockfile, pbr, funcsigs, mock, mbed-greentea, beautifulsoup4, fuzzywuzzy
  Found existing installation: setuptools 18.5
    Uninstalling setuptools-18.5:
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/commands/install.py", line 342, in run
    prefix=options.prefix_path,
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/req/req_set.py", line 778, in install
    requirement.uninstall(auto_confirm=True)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/req/req_install.py", line 754, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/req/req_uninstall.py", line 115, in remove
    renames(path, new_path)
  File "/Library/Python/2.7/site-packages/pip-9.0.1-py2.7.egg/pip/utils/__init__.py", line 267, in renames
    shutil.move(old, new)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
    raise Error, errors
Error: [('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py', "[Errno 1] Operation not permitted: '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py', "[Errno 1] Operation not permitted: '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/markers.py'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc', "[Errno 1] Operation not permitted: '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib/__init__.pyc'"), ('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib', "[Errno 1] Operation not permitted: '/tmp/pip-WtQCOE-uninstall/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python/_markerlib'")]
```

@screamerbg

janjongboom, 2018-01-04:
You can fix it with:

```
pip install -r requirements.txt --user python
```

screamerbg, 2018-01-12:
@theotherjimmy, @bridadan On Windows --user python would fail miserably as it tries to find user profile 'python'. On Mac this seems to be the only way to go. Not sure how this works on Linux, but Linux doesn't have "protected" libraries, like Mac ElCapitan/Sierra/HighSierra, therefore 'sudo' should work just fine. Any ideas how this could be handled?

theotherjimmy, 2018-01-12:
@screamerbg What exactly are we handling? Cross platform `pip install`?

screamerbg, 2018-05-26:
@AnotherButler Could you please check with @janjongboom about some sort of cross platform `pip install` section in the documentation?

It seems that we have 3 cases:
* Windows - `pip install` works just fine
* Linux - `pip install` may work fine, but often it requires `sudo`, e.g. `sudo pip install`
* Mac - `pip install` often doesn't work, `sudo pip install` works well for manually installed Python, but the system Python requires `pip install --user python`, which fails miserably on Windows for example.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2019-08-22 by ciarmcom