ARMmbed/mbed-cli #599: Docker image for development

davidhyman, 2018-01-10:
Development on such a project without a Docker image will be pot luck.

screamerbg, 2018-01-10:
Where this requirement is coming from? Do you know how many embedded engineers use Docker? :)

davidhyman, 2018-01-11:
Maybe embedded engineers don't, and we're not about to change that by adding one to mbed-cli (although its certainly possible to make serial connections to connected boards from inside a container).

My primary motivation for this would be to have a reproducible build / test environment. Specifically, there are currently ~70 lines of tests run on CircleCI that aren't reproducible in a stable environment locally.

davidhyman, 2018-01-11:
For reference: https://github.com/ARMmbed/mbed-tools-docker/blob/master/Dockerfile

loverdeg-ep, 2018-01-11:
@screamerbg Maybe more embedded engineers than you'd think...
Modernization of the embedded development process is a top 2 reason I'm pushing my peers to look into mbed.

screamerbg, 2018-01-12:
@loverdeg-ep @davidhyman Fair enough. In the same time let's make sure that Docker isn't required to develop mbed OS, nor mbed OS tools (including mbed CLI)

aschmidt75, 2018-03-02:
I'd love to see standard build envs for mbed. 

I'm using https://github.com/aschmidt75/docker-mbed-cli-gcc-arm
but would appreciate a way supported by ARM.

loverdeg-ep, 2018-03-02:
@aschmidt75 are you attempting to take the Eclipse Che path?

aschmidt75, 2018-03-03:
I don't know about Eclipse's Che path, i'm just using containers for years now for this kind of build system helpers - works quite well.

jaimejim, 2018-08-15:
adding to @loverdeg-ep and @davidhyman , on the IETF we are using the mbed cli to test what we call a [suit manifest file](https://tools.ietf.org/html/draft-moran-suit-manifest-02). There are [detailed notes](https://git.io/fNYC6) about a couple of hackathons we have had and a common problem we identified is that newcomers need a lot of time to get started with their environment. We found it could be very useful to have an isolated environment using docker or nix.

theotherjimmy, 2018-08-15:
Thanks for the link @jaimejim 

@screamerbg Requirement or not, we certainly have proof that it would be used.

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

loverdeg-ep, 2018-11-21:
@BlackstoneEngineering please review my comments regarding Eclipse Che as well.

Our path internally is chasing Eclipse Che. It would provide a similar, even better solution than the Cloud 9 path that was taken in the past.

I want devs to be able to go as far as remote step debugging without installing a single tool on their computer.

Also seems like a natural progression for the work @thegecko is doing with Theia.

BlackstoneEngineering, 2018-11-22:
@SenRamakri / @theotherjimmy , i'm seeing a demand for a docker container from customers and developers in the wild. As we are about to engage in general installer updates / rework I would like to add getting an official mbed docker container with all the tools and compiler in it added to the list.

markus-becker-tridonic-com, 2018-11-23:
An official Docker image would also help with running a CI for mbed based products.

trowbridgec, 2018-11-29:
I agree with @loverdeg-ep and company that this would be an awesome tool to have!

loverdeg-ep, 2018-11-29:
> help with running a CI for mbed based products.

My thoughts exactly.

jaimejim, 2019-03-23:
~~~
docker pull jaim/mbed-cli
~~~

Could be useful for those who want to give it a try. The Dockerfile is on the repo.

40Grit, 2019-03-25:
@jaimejim you find a way to use the gdb client in the container for debug with an IDE running on the host machine?

loverdeg-ep, 2019-04-26:
https://www.arm.com/company/news/2019/04/docker-and-arm-partnership

We have our own but still waiting for something official...

closed 2019-10-31 by ciarmcom