ARMmbed/mbed-cli #601: support Python3

davidhyman, 2018-01-10:
#376 updates documentation such that 'Python 3 is not supported'

This ticket is to add Python3 support.

Changes required:
- All print statements
- Some Lambdas e.g. L1118 `sorted_scms = sorted(sorted_scms, key=lambda m__: not m__[0]) `
- Some iterators

screamerbg, 2018-01-10:
mbed CLI is the last tool to update from the mbed OS tools. The following tools are Python2 compatible and needs updating first:
* mbed OS build system (under mbed-os/tools/)
* mbed OS exporters (under mbed-os/tools/)
* mbed OS test spec generator (under mbed-os/tools/)
* the rest of mbed OS tools
* mbed greentea (https://github.com/ARMmbed/greentea)
* mbed htrun (https://github.com/ARMmbed/htrun, called by greentea)
* mbed ls (https://github.com/ARMmbed/mbed-ls)

mbed CLI depends on these python tools to function correctly, therefore claiming that mbed CLI supports Python3, but getting failures in any of these tools due to Python3 incompatibility won't work.

screamerbg, 2018-05-26:
Python 3 support is now available as of Mbed CLI 1.7. Thanks for your contribution!

closed 2018-05-26 by screamerbg