ARMmbed/mbed-cli #602: test coverage

davidhyman, 2018-01-10:
It would be useful to investigate the level of test coverage as this indicates how much confidence there can be in making changes. Considering the number of branches in the code I imagine it will be quite low.

https://codeclimate.com/github/davidhyman/mbed-cli

This would indicate other problems in the structuring and size of functions used.

theotherjimmy, 2018-01-11:
@davidhyman We use coverage + coveralls for the Mbed OS build tools, Mbed Greentea, Mbed Htrun and Mbed ls. You are welcome to set something similar up for Mbed CLI. Before changing any test code, I would first like to see what the coverage looks like.

screamerbg, 2018-01-12:
@davidhyman There's a lot going on circleyml that doesn't seem to be taken into account in these stats, e.g.
* Importing with Git, Hg, Bld
* Updating to a specific version with Git, HG, Bld
* Setting up config variables, default target
* Listing of supported toolchains, targets for compile, export, etc.

Do you could look at translating some of these to py.test tests so they are accounted against the coverage statistics you mentioned?

theotherjimmy, 2018-01-12:
@screamerbg after #608, We can use `coverage -m mbed` instead of `mbed` on the command line, and get the coverage info for those commands. That sounds like a bit less work.

davidhyman, 2018-01-12:
Yes - in theory - it would be a matter of using `coverage run mbed ...`  in place of `mbed` (then `coverage combine` and some other invocations)
In practise it may be somewhat fiddly getting this going, if we're only able to debug through CircleCI.

davidhyman, 2018-01-12:
converting them to a suite of pytest tests with subprocess calls could also work though, and provides a single entrypoint for all tests

adbridge, 2018-10-02:
*Issue header has missing or incorrectly filled in fields! Please fix.*

closed 2019-10-31 by ciarmcom