ARMmbed/mbed-cli #604: undocumented environment vulnerability

davidhyman, 2018-01-10:
L48:
```
if os.path.basename(python_cmd).startswith('mbed'):
    python_cmd = 'python'
```
The validity of doing this should be documented, because as it stands, it appears wrong: 'python' will invoke whatever executable has that name on the current system path. It may not be the same interpreter that is executing the current file. It may not work with other aliases of python, or in environments where python is not on the system path. This could cause a number of issues e.g. when using a python interpreter that is directly invoked from an absolute path, or when using virtual environments.

screamerbg, 2018-01-10:
Mbed CLI executes a number of external tools:
* Python (+ all tools in mbed-os/tools/)
* Git
* Mercurial
* PIP
* mbed greentea

Contributions to the documentation are welcome.

davidhyman, 2018-01-11:
By documentation, I meant in the code, to explain this decision if it is explicitly intended to behave in the current manner.

What I would recommend instead, is using `sys.executable` in all cases where Python is being invoked. Alternatively, we could take the python executable path as a configurable argument?

For Pip, we should use `{sys.executable} -m pip` which would ensure the same executable as used by `import pip` on L1389. Alternatively, the subprocess call to Pip install could be replaced by code that uses the imported pip to achieve the same goal (or the `import pip` could be replaced with a subprocess call) for consistency.

It is also worth considering whether use of [pipenv](https://github.com/pypa/pipenv) would be useful for the python/pip operations (It may not be appropriate if the user is already inside a venv when executing mbed-cli).

screamerbg, 2018-01-11:
Thanks for the clarifications. Your suggestion makes sense. To make this more efficient in the future, perhaps add more detailed description so the intent/feedback/recommendation is clearer.

Regarding pipenv, it's a requirement to mbed CLI is to support both pipenv and native environments, without changing the environment on its own. In your experience, does `{sys.executable} -m pip` take into account the current location and behave the same as executing `pip` when inside a virtual environment? I assume yes, but I haven't used `pip` in the way you propose.

theotherjimmy, 2018-01-11:
@screamerbg I'm pretty sure that `{sys.executable} -m pip` invokes the same code as `pip` with the added guarantee that it uses the same python interpreter as the executing code, matching versions.

screamerbg, 2018-02-06:
This is now merged and released in Mbed CLI 1.4.0

closed 2018-02-14 by screamerbg